raspberrypi/noobs #534: Q: Memory Usage

procount, 2019-03-12:
Is NOOBS restricted on how much RAM it will use (e.g. to be compatible with all RPi models)?

The output of `free` (on PINN) gives me:
```
# free
             total         used         free       shared      buffers
Mem:        233464       225692         7772          428        89200
-/+ buffers:             136492        96972
Swap:            0            0            0
```
and output of `cat /proc/meminfo` yields:
```
# cat /proc/meminfo
MemTotal:         233464 kB
MemFree:            6588 kB
MemAvailable:     171772 kB
Buffers:           85356 kB
Cached:            73892 kB
SwapCached:            0 kB
Active:            84736 kB
Inactive:          83160 kB
Active(anon):       8852 kB
Inactive(anon):      256 kB
Active(file):      75884 kB
Inactive(file):    82904 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:              5200 kB
Writeback:             0 kB
AnonPages:          8680 kB
Mapped:             4576 kB
Shmem:               428 kB
Slab:              19892 kB
SReclaimable:      13408 kB
SUnreclaim:         6484 kB
KernelStack:         728 kB
PageTables:          384 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      116732 kB
Committed_AS:      76700 kB
VmallocTotal:    1835008 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
CmaTotal:           8192 kB
CmaFree:              12 kB
```
Which looks to me like there is only 233 MB or RAM available. An RPi 3B has 1GB of RAM, so where is all the rest of the memory being used? Even accounting for the tmpfs, initramdisk and any GPU mem, it still seems to be lacking.  Or am I missing something?

If it is restricted, is it possible to enable full use of the RAM?

lurch, 2019-03-12:
This is probably a question for @ghollingworth or @popcornmix :wink:

procount, 2019-03-12:
I expected it would just be a kernel config parameter somewhere... 🤷‍♂️ ?

maxnet, 2019-03-12:
>If it is restricted, is it possible to enable full use of the RAM?

Delete recovery.elf, use normal start.elf fixup.dat

procount, 2019-03-12:
Thanks I'll give it a go.

procount, 2019-03-12:
I used start.elf and fixup.dat from rpi-firmware-fbad6408c4596d3d671736ee0571aae444f24e68.
It ended with a kernel panic - unable to mount root fs on unknown-block(179,2) 😢

maxnet, 2019-03-12:
>It ended with a kernel panic - unable to mount root fs on unknown-block(179,2)

Probably need a config.txt with:

```
initramfs recovery.rfs
```

procount, 2019-03-12:
No difference. I even added `followkernel` to no effect.

_(I assume you are taking account of your - admittedly very old - mod:
" initramfs -> squahsfs initrd to save memory")_

maxnet, 2019-03-12:
>(I assume you are taking account of your - admittedly very old - mod:
>" initramfs -> squahsfs initrd to save memory")

Ah, probably need to rename recovery.cmdline to cmdline.txt for that then.

Guessing recovery.cmdline is only consulted if recovery.rfs exists or something like that.
But do not know for sure (firmware is a big black box for me, do not have source code)

procount, 2019-03-12:
(Kicking myself repeatedly) I should have guessed that was required as soon as I replaced recovery.elf with start.elf.
It now boots!
That's more like it. Free reports 952MB total memory.

I'm currently using the RPF touchscreen, and I'm now noticing a black line scrolling down the screen at a rate of about 2Hz that wasn't there before. Will revert to recovery.elf and test it's not a loose ribbon or other issue.

procount, 2019-03-12:
I think it's time for a new ribbon cable. 
Many thanks, @maxnet. (I'm hoping this will speed up OS backups)

closed 2019-03-12 by procount

lurch, 2019-03-12:
The firmware is also a big black box to me, but https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained#noobs-bootup-low-level might be helpful :slightly_smiling_face:

procount, 2019-03-12:
I might have to try it again to verify, but I think if recovery.elf and start.elf are both present, then recovery.elf will take precedence. In which case step 2 might need a minor mod to be accurate.

(my backup isn't going much faster, though 😢)

lurch, 2019-03-12:
The backup will probably be limited by the write-speed of wherever you're writing to, which will likely dwarf any other factors?

procount, 2019-03-12:
We're wandering off topic, (but as I closed it anyway :wink:)  consider that I am just doing the steps of "creating a custom OS", but doing it on the RPi, not on a PC. 

When we install an OS, we decompress/expand the files, so we are writing more than we are reading.
When I backup an OS, I am compressing the files, so I am reading more than I am writing. 

Given that reading is generally much faster than writing, and ignoring the compression/decompression delays for a minute, It should be faster to backup than install, because there is much less writing. But once you take the compression into account, it is 4-5 times slower, leading me to believe that the compression is the main pinch point. 

When I first tried this, using xz -9 was out of the question due to lack of memory (maybe that will improve if I now use 1GB of RAM?). But even using the default xz compression took a long time, hence my switch to gzip, and also to ensure it would work on the lower memory/speed models.
I had hoped that the extra memory might have improved the gzip compression, but initial results do not indicate that. I'm probably just hitting the limits of the RPi and USB bandwidths.

FYI, I was using a 128GB Sandisk Ultra USB 3.0 memory stick formatted as NTFS to write to, and reading from a 64GB Sandisk Ultra uSD card

lurch, 2019-03-12:
Continuing the off-topic discussion.... :wink: 
Yes, any kind of compression on the Pi will always be *much* slower than on a PC. IIRC gzip has different compression levels? You might want to experiment e.g. under Raspbian to see what kind of compression speeds you get from different compression levels.
Also, it might be worth reformatting your memory stick as ext4, to see if Linux can write to that faster than NTFS? Again, I'd advise benchmarking under Raspbian as that'll be much easier than trying to test things within PINN :-)

procount, 2019-03-14:
@lurch - pigz to the rescue! Backs up an OS almost as quick as installing it. Quite a stress test on the old 3B processor though. Measured 81.7 degrees with the thermal warning on the screen. 3B+ will probably fare better. Need to try PiZero too.

carlonluca, 2021-01-27:
Hello,
sorry for writing to this old issue, but I have a question related to this. What is the downside of running NOOBS with start.elf and fixup.dat? Why is NOOBS using recovery.elf instead of start.elf?
Thanks!

lurch, 2021-01-27:
Have a look at the link I posted earlier https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained#noobs-bootup-low-level
I dunno the full details, but I believe "recovery mode" takes a few shortcuts e.g. it doesn't need a `config.txt` file.

carlonluca, 2021-01-27:
Yes, I read it, but still I cannot clearly see the real advantage. I read this cryptic part: "switches the firmware into "NOOBS mode"". Not sure what "NOOBS mode" is.
xz requires a lot of RAM, so the missing portion in NOOBS is pretty relevant.
Thank you for your answer.

lurch, 2021-01-28:
> Not sure what "NOOBS mode" is.

Where I say "NOOBS mode" in that wiki-page I'm actually referring to "recovery mode" (i.e. they're the same thing). I can't be sure (since it was developed before I started looking at NOOBS), but I _think_ the aim of recovery-mode might have been to minimise the number of files required on the FAT partition of an SD card? e.g. there's no config.txt or fixup.dat files. There might be other differences, but any further details will require input from @XECDesign or @popcornmix . And of course NOOBS and recovery-mode where developed before device-trees (and different kernels) became a thing :wink: (i.e. there's a lot fewer files in https://downloads.raspberrypi.org/NOOBS_lite/images/NOOBS_lite-2013-11-02/NOOBS_lite_v1_3_2.zip than there are in https://downloads.raspberrypi.org/NOOBS_lite/images/NOOBS_lite-2020-09-15/NOOBS_lite_v3_5.zip )

> xz requires a lot of RAM, so the missing portion in NOOBS is pretty relevant.

Can you illustrate an example where the "lack of RAM" in NOOBS is causing a problem with installing an OS? It's been a *long* time since I looked at xz in detail, but I think I remember reading that one of its features is that it only needs a fixed amount of memory for decompression?

procount, 2021-01-28:
recovery.elf was the same as start_cd.elf up until recently.
With the latest build environment changes, it now uses start.elf and fixup.dat, so you get the full memory available in the latest version.
Whilst config.txt isn't normally necessary, it can still be used to set specific video modes, for instance, where this is necessary.

carlonluca, 2021-01-28:
> Can you illustrate an example where the "lack of RAM" in NOOBS is causing a problem with installing an OS? It's been a _long_ time since I looked at xz in detail, but I think I remember reading that one of its features is that it only needs a fixed amount of memory for decompression?

This explains how xz uses memory: https://linux.die.net/man/1/xz. My understanding is that it may be useful to have the entire memory available, which is why I'm trying to understand the downside. What I see is that uncompressing my image fills the available memory entirely. Nevertheless, I have no confirmation of a case in which it failed because of insufficient RAM, and this is what I see after the modifications above:

```
# cat /proc/meminfo
MemTotal:         996820 kB
MemFree:            6836 kB
MemAvailable:     842868 kB
Buffers:           34876 kB
Cached:           771704 kB
SwapCached:            0 kB
Active:           135212 kB
Inactive:         753984 kB
Active(anon):      82856 kB
Inactive(anon):       36 kB
Active(file):      52356 kB
Inactive(file):   753948 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:             75188 kB
Writeback:           336 kB
AnonPages:         82708 kB
Mapped:            16032 kB
Shmem:               184 kB
Slab:              53036 kB
SReclaimable:      45348 kB
SUnreclaim:         7688 kB
KernelStack:         720 kB
PageTables:          460 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      498408 kB
Committed_AS:     121740 kB
VmallocTotal:    1064960 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
Percpu:              304 kB
CmaTotal:           8192 kB
CmaFree:               0 kB
# free -m
             total         used         free       shared      buffers
Mem:           973          965            8            0           37
-/+ buffers:                927           45
Swap:            0            0            0
```

This is a bit weird, as the doc I linked states: "the default limit is 40 % of total physical RAM", but also "worst-case memory usage of the decompressor is several gigabytes". My image succeeds with the current 230MB limit, but fills 100% 1GB when available. This is a bit confusing. In any case, I was wondering what advantage there could be in keeping the current limitation.

carlonluca, 2021-01-28:
> With the latest build environment changes, it now uses start.elf and fixup.dat, so you get the full memory available in the latest version.

Are you saying that building NOOBS today would result in using start.elf and fixup.dat, giving access to the entire GB available?
Thank you for your answers.

maxnet, 2021-01-28:
>My image succeeds with the current 230MB limit, but fills 100% 1GB when available. 

It is normal that Linux uses any spare memory as disk cache.
770 MB in your case.

>Cached:           771704 kB

It can free that again if any application needs it.

procount, 2021-01-28:
> Are you saying that building NOOBS today would result in using start.elf and fixup.dat, giving access to the entire GB available?
Thank you for your answers.

Yes.

carlonluca, 2021-01-28:
Ah yes, maybe here: 00822079b514e5db9dbd6e0e3c86628b94b4fa9a.
Thanks for the info!

carlonluca, 2021-01-28:
> It is normal that Linux uses any spare memory as disk cache.
> 770 MB in your case.

You are right, I was confused by the +/- buffer value returned by free. Anyway, still pretty near to what NOOBS could provide. But I understand this is not a problem anymore with more recent versions. Thanks!