raspberrypi/noobs #535: Raplace background image.

pcs3rd, 2019-03-16:
Is there a feature or way to change the image shown at boot?

XECDesign, 2019-03-16:
Replace https://github.com/raspberrypi/noobs/blob/master/recovery/wallpaper.png with a file of your choice and rebuild.

lurch, 2019-03-16:
And edit this line (and rebuild) if you want to change the background colour (that gets displayed outside of the image)
https://github.com/raspberrypi/noobs/blob/master/recovery/config.h#L9

procount, 2019-03-16:
Or just switch to PINN which already has user selectable background colour and image :wink:

pcs3rd, 2019-03-18:
If I switch to PINN, can i keep my existing raspbian?

procount, 2019-03-18:
Yes, but as always, _**back up your SD card first**_ (just in case you make a mistake).

You just need to overwrite your NOOBS partition with the contents of pinn-lite.zip.
_**BUT, YOU MUST EDIT `recovery.cmdline` TO REMOVE 'runinstaller'**_ otherwise the whole of your SD card will be wiped (hence the need to back it up first!) :wink:

If you have a spare SD card, you can practice beforehand by installing NOOBS then Raspbian, then replacing NOOBS with PINN  whilst removing runinstaller from recovery.cmdline before rebooting to check Raspbian is still operational. (This does not remove the need to backup your original SD card first!).

closed 2019-11-19 by pcs3rd