raspberrypi/noobs #536: cannot get UART interface to work with RPI3B+

cokron, 2019-03-22:
Hello everybody,
thank you for having created NOOBS!

I have been using it since 3 years. A client of mine builds devices that are a combination of a Raspberry PI and their custom hardware, attached via the UART interface. The custom hardware has an LED that displays the status of the device.  _The LED is talked to via UART_ – not by switching on an output PIN, but by sending a command via UART. 

A user can reset the device, in which case, NOOBS reinstalls the system. 
The LED should indicate that NOOBS is reinstalling.

This all works fine with RPI1 and 2 and an earlier version of NOOBS, where I was able to simply write to `/dev/ttyAMA0`. 

With the current NOOBS and RPI3B+, writing to `ttyAMA0` or `ttyS0` has no effect.

What do I need to do to enable UART either ttyAMA0 or ttyS0?

Btw, I write the command to set the LED by writing to ttyAMA0 or ttyS0:

```
echo -e "\xa5\x86\x04\x02\xff\xff\xff\xe1\xd3" > /dev/ttyAMA0
```

This is run from `buildroot/package/recovery/init`.

I am aware that for RPI3B+ the ttyAMA0 is now used by Bluetooth. So I tried writing to `/dev/ttyS0` or `/dev/ttySerial0`. This does not work.

I then added the `dtoverlay=pi3-disable-bt` to `recovery.cmdline` and tried writing to `/dev/ttyAMA0`. Nothing happens. 

My recovery.cmdline looks like this:

```
runinstaller dtoverlay=pi3-disable-bt quiet enable_uart=1  ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 coherent_pool=6M  elevator=deadline silentinstall gpiotriggerenable
```

I removed any `console=....` entries.
The baud rate is set correctly (according to stty which I executed in a terminal when NOOBS started).

I there anything else I need to do to get communication with the UART interface working?

Any help is greatly appreciated! I tried a full day and need to get the LED via UART working.

Thank you!

cokron, 2019-03-22:
PS: UART works fine on the installed OS (Raspbian), once the RPI3 rebooted into the fresh system.

lurch, 2019-03-22:
Does https://www.raspberrypi.org/documentation/configuration/uart.md help? Although obviously there'll be some differences between the "full Linux" of Raspbian and the "cutdown Linux" buildroot used by NOOBS.

cokron, 2019-03-22:
Hello @lurch Andrew, 
thanks a lot for taking time to help me and pointing me to that link!

* I had tried disabling `pi3-disable-bt`, but was unaware that I might have to stop the hciuart service. I'll try that. (any advise on how to compile NOOBS so that it does not start hciuart – if it does start this at all?)
* I will also  try _without_ `pi3-disable-bt` so that UART remains at `/dev/ttyS0` and set `core_freq=250`. Something I was unaware of until now. 

What puzzles me is that in the `cmdline.txt` of the OS that gets installed (Raspbian), I did not have to add `core_freq=250`.

I'll post here any insights I'll get on my way.

procount, 2019-03-22:
As far as NOOBS is concerned, the device-tree stuff is just about configuring the hardware: routing the serial ports and setting the appropriate clock frequency. There are no hciuart (bluetooth) services within NOOBS to worry about (In fact it's something that I'd like to add in order to get a bluetooth keyboard working, but that's another story!)

lurch, 2019-03-22:
I dunno if it was a typo in your earlier comment, but IIRC the `dtoverlay` stuff needs to to go in `config.txt` rather than `recovery.cmdline`? And in Raspbian there's `dev/serial0` and `/dev/serial1` aliases you can use, but I dunno if these exist in NOOBS.

procount, 2019-03-22:
I suspect that might be the problem - it's not just in his comment, but displayed in his paste of recovery.cmdline.

In my PINN (should be the same) I can see /dev/ttyAMA0 and /dev/tty50.
No /dev/serialX device.

lurch, 2019-03-22:
`/dev/tty50` - LOL, did you mean `/dev/ttyS0` :wink:

procount, 2019-03-22:
The problem with tiny purple letters on a black screen, is I can't read them properly! :wink:
So actually I only have ttyAMA0 with my default setup.

lurch, 2019-03-22:
If only you had a UART connection that you could copy'n'paste from... :rofl:

procount, 2019-03-22:
🤣 and I was working over SSH, so I could have copied it!