raspberrypi/noobs #537: Updating NOOBS

Nickduino, 2019-03-22:
I'm going to setup a new Raspberry Pi 3B+

I'd like to use and old SD with NOOBS preinstalled I got with my first Pi (2014 I think). I cannot really manipulate my SD, though (it was working fine on my old Pi just a few weeks ago): I can't access all partitions, I can't format it, it seems to be write protected...

[image]

What would you advise me to do to regain access to my sdcard?

XECDesign, 2019-03-22:
If SD Card Formatter is having trouble clearing it, it's not a NOOBS issue. The problem could be with the reader or the card itself. I would recommend asking the question on the forum.

Nickduino, 2019-03-22:
> If SD Card Formatter is having trouble clearing it, it's not a NOOBS issue.

But you could be used to the fact that the official sd cards sold by the Raspberry foundation in 2014 had this or that limitation

XECDesign, 2019-03-22:
If you use windows' own partitioning tool to remove all the partitions on the sd card before running the formatter, is the behaviour any different? Also, try quick format. There's rarely a reason to overwrite everything.

Nickduino, 2019-03-22:
I can see the partitions through the windows tool but I cannot remove the 5.88 MB one.

Quick format: the tool tells me everything went ok...but I still see the files in the 1.43 MB partition (the only one openable in the explorer)

XECDesign, 2019-03-22:
I'd try it with gparted from linux, since it should at least give you a more meaningful error. An ubuntu livecd will work, if you don't have an install handy.

lurch, 2019-03-22:
It's also worth bearing in mind that all flash-based media (including SD cards) eventually wears out, so if you've been using your SD card regularly for the last 5 years it's possible there are sectors on your card which can no longer be erased?  (I can't think of many other reasons why formatting would suddenly fail at 19%)

You might be surprised to see just how much cheaper SD cards are now than they were in 2014... :wink:

Nickduino, 2019-03-22:
> (I can't think of many other reasons why formatting would suddenly fail at 19%)

I can think of a good reason: the formater thinks it's erasing the first partition, which is about 1.5 BG out of 8 (about 20%) and fucks up on the second partition:
[image]

I can see the first partition content in the explorer and I can **seemingly** write and delete stuff on it...but they reappear or disappear within a few seconds.

I'm pretty sure I have one small recovery partition and one big "system" partition, I can't modify the first nor see the latter

XECDesign, 2019-03-22:
You're describing a bad sd card. When they fail, they go into a read-only mode. They accept writes, as far as the operating system is concerned, but nothing actually changes. Then when the cache is cleared, the OS reads the original files again.

closed 2019-03-22 by XECDesign