raspberrypi/noobs #538: new 16Gb noobs wont install on pi3+

pearcecg, 2019-03-26:
The latest 16Gb preinstalled noobs card wont install Riscos.
Error says : cant install RiscOs - recovery partition too large.
This is with latest 3B+, purchased at the same time.

The Pi Hut less than helpful, suggesting I contact the software developer.
I have several Pis, including an earlier 3B, and have never had this sort of problem.
I thought noobs was supposed to be painless.

Help appreciated,

Charles Pearce

procount, 2019-03-26:
Hi Charles,

I already provided you with an answer on your forum thread: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=236448#p1446123

The problem is that the total size of the Raspbian FULL and LibreELEC OS distribution files on the recovery partition overlap with the location where RISCOS expects to find its binary blob (file tables), so RISC OS cannot be installed.

If you want to install RISCOS via NOOBS, then I suggest you convert your SD card to the NOOBS LITE version:
1. delete the /os folder from your NOOBS SD card to remove the Raspbian FULL and LibreELEC OS distributiuon files.
2. Edit recovery.cmdline to add 'runinstaller' as an option to the first (and only) line of that text file. 
3. Boot up NOOBS which will now repartition your SD card and remove any existing OSes you have already installed.
4. Connect your RPi to the internet via Ethernet or set up your wifi to connect to your wireless router.
5. Select RISCOS and install it.

Alternatively, if you really want the Raspbian distro files on your SD card, you could in step 1 perhaps replace Raspbian FULL with the normal Raspbian version from here -> http://downloads.raspberrypi.org/raspbian/

procount, 2019-03-28:
documentation updated (#539)

lurch, 2020-06-12:
Closing, as #539 has been merged.

closed 2020-06-12 by lurch