raspberrypi/noobs #541: questions upon reading readme

bjb, 2019-04-14:
Questions I still have upon reading https://github.com/raspberrypi/noobs (the README).

Note that the user forum says you should not use NOOBS.  But I think NOOBS is the raspberry pi supported approach, so here are my questions.

1)  It says "on first boot of NOOBS", do you mean the first time you run it?  Or does it actually have to boot on the raspberry pi?  I thought it was software you run on your other machine (let's call it the "dev machine") to prepare an SD card to boot the pi with?

2)  It says "on first boot of NOOBS it will format your SD card ... "  but if NOOBS is already on the SD card, then the SD card should already be formatted?  If it is running on the pi, and it formats the SD card on boot, won't that make the system crash?  Does it require two SD cards being plugged in?  My pi (admittedly on old one) doesn't have two SD card slots.

If you are actually running NOOBS on your dev machine then will it go around your dev machine devices, looking for SD cards (and maybe USB drives) to format?  If it boots on the pi and then formats the SD card, how is it going to find the OSs to be able to install them?  If it is running on your dev machine, hopefully it will ask for permission before proceeding to do formatting?

3)  The NOOBS software makes a list of OS's to install ... pulling from my dev machine?  or the raspberry pi? and from the web.  It will only install the latest version of any OS.

So if I have OS "strudel" (made up name for illustrative purposes), version 2 and 4 on my dev machine and version 3 and 5 on the web, will it refuse to install 4 onto the SD card until I break the network connection?

Is it going to install onto the SD card or "onto the pi"?  Can I install versions 4 and 5 of "strudel" onto the SD card?

Do I really have to make a directory off the root of my dev machine called "/os", for the use of NOOBS?  If not, do I have to run NOOBS as root, so it can make that directory?

Or, maybe there is a directory on the SD card called /os ... are there multiple subdirs there, one for each installed OS?  It would be good to give an overview of what we're aiming for.  (I downloaded NOOBS lite, and there is an /os directory with nothing in it, which is fine ... but I'm not sure what I'm aiming for and I didn't want to install two OS's that I'm not going to use - raspbian and the elec thing).

If the situation we want to have is an SD card that boots NOOBS and some OS's, and we're going to take one of those OS's and replace the whole SD card contents with that one OS, then it would be good to say so ... otherwise it is easy to get confused reading these instructions trying to divine what is supposed to be happening.  As can be seen by my (possibly ridiculous) guesses on what is supposed to be happening.

I would be happy to supply patches to the README but I really don't know (yet) what it should say.

bjb, 2019-04-14:
Ok I haven't finished looking at these, but they seem like a good way to describe NOOBS ...
http://qdosmsq.dunbar-it.co.uk/blog/2013/06/noobs-for-raspberry-pi/
https://www.youtube.com/watch?v=TyFDaMpdh2c

After I study these and try it out, maybe I will try to make patches for README.md.  This task may run to next weekend though and next weekend is long weekend, I will be busy with family, don't be alarmed if it takes a few weeks before you hear from me again.

procount, 2019-04-14:
Hi there. Sorry you found the instructions for NOOBS so confusing. It's really not meant to be so hard. In fact it's supposed to be "New Out Of the Box" software to get newbies started. It was developed many years ago when the alternatives to flashing an SD card with a Linux Operating System were harder, particularly when the vast majority of new users were used to Windows. That situation has now changed and tools like Etcher make flashing an SD card with an OS a lot easier. This is why some Forum users suggest not to use NOOBS. Nevertheless, I think NOOBS still has a place, mainly because it gives you a choice of different operating systems to install with a point and click interface, and allows you to install several on your SD card at once, creating a "multi-boot" SD card. I liked it so much that I forked my own version called PINN which includes a much wider choice of operating systems to choose from, and many additional features.

The best way to understand NOOBS is to just use it, so I think it's best to correct your viewpoint so you can understand how to use the instructions to get NOOBS up an running. I'm sure any improvements to the README which will help explain NOOBS will be welcomed, but first you need to have a thorough understanding.

I'm going to assume that your 'dev' machine is a Windows PC. 
The first 3 steps in the [setup](https://github.com/raspberrypi/noobs#setup) paragraph are to be performed on your Windows Dev PC. Download the noobs.zip or noobs-lite.zip file from the internet, unzip it, and copy the contents onto your FAT32 formatted SD card in an SD card slot, or better in a USB SDcard reader. 

Once you have NOOBS on the SD card, you can remove it from the PC and insert it into your Raspberry Pi. NOOBS consists of a small Linux OS and a small recovery program that will only run on the Raspberry Pi. It will not run on your Windows Dev machine.

When you turn on you Raspberry Pi with NOOBS on the SD card, it will firstly shrink its partition to a minimum size and re-partition the rest of the SD card to be prepared to install the new operating systems on to it. This will tak a little while, but then you will be presented with the menu of operating systems that you can install.

NOOBS will source its operating systems from 3 main places:
1) If you have installed the Full NOOBS, it will include some OSes on the SD card including Raspbian and LibreELEC in the /os folder (each OS in a folder of its own). NOOBS-lite is the same as NOOBS except it does not include any OSes in the /os folder, so it takes up less of your SD card space (some people think this 1.8GB is wasted space, so use noobs-lite if this is your viewpoint)
2) If you plug your raspberry Pi into a network (or use wifi) then NOOBS can install an OS from the internet. The list of OSes available from the internet is much bigger so it gives you a wider choice.
3) NOOBS can also install an OS from the /os folder of a USB drive. But you have to copy the OS onto the USB driver manually in the first place.

Hopefully that explains the process sufficiently to get you started.
When I developed PINN, I wrote a quick-start guide to explain the process (which is the same for NOOBS or PINN). Maybe have a read of that to see if it explains it any better -> https://github.com/procount/pinn#quick-start

procount, 2019-04-14:
Regarding the links you posted, please bear in mind that they are quite old now, although the same basic process still applies. Also the first is for Linux users, whilst the second is for Windows users. What type of "dev" machine do you use?

Just a point to bear in mind - NOOBS does not install standard image files, but requires the OS installation files to be formatted specifically for NOOBS (these are much smaller compressed files so they take up the minimum amount of space on the SD card for FULL NOOBS and can be downloaded quickly). So your choice is a little limited from the official list. However, there are over 60 other OS variants that have been converted for PINN, most of which can be used by NOOBS too.

Regarding your example of the "Strudel" OS and versions, it just means that if the same Strudel OS is available from multiple sources (SD card, internet or USB drive), that only the source with the most recent date will be listed.

lurch, 2019-04-14:
There's also more detailed setup info at https://projects.raspberrypi.org/en/projects/raspberry-pi-setting-up/3 and https://www.raspberrypi.org/documentation/installation/noobs.md

And I've just modified the README to say "On first boot NOOBS will repartition your SD card..." :-)