raspberrypi/noobs #542: Supported USB WiFi devices for Recovery Mode

allenhuffman, 2019-05-06:
The README mentions the official Pi USB WiFi module, but that was apparently discontinued in 2018. What are the requirements for a USB WiFi module for Recovery Mode?

XECDesign, 2019-05-07:
I don't know if there are any usb devices currently sold which use a supported chipset.

Devices which use the brcmfmac driver should be covered. That includes the on-board wifi of pis which have it.

lurch, 2019-05-07:
@procount Did you add support for a wider range of WiFi dongles to PINN ?

procount, 2019-05-07:
Yes, but I had to drop a lot of Realtek drivers because of the maintenance of having to resolve name-clashes (They all use the same driver source code). I will probably put them back in when I support modular drivers.

But I have already answered Allen on the Forum: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=239630 . He has a Ralink RT5370 which is supported by PINN.

closed 2019-05-07 by XECDesign