raspberrypi/noobs #543: Query: ext4 meta data size calculation error?

procount, 2019-05-20:
https://github.com/raspberrypi/noobs/blob/54ac40f7c59e945d5a1389e356c4531b53ca0216/recovery/multiimagewritethread.cpp#L99

I think the above line is supposed to add 1% overhead in size to account for the ext4 metadata. But it seems to be adding 1% of the **accumulated total** of all the partitions so far encountered. Shouldn't it be 1% of the nominal size of **THIS** partition only?

lurch, 2019-05-20:
Oooh, well spotted. Given that the code is inside a `foreach` loop, I _suspect_ that it was supposed to be:
```c++
    totaluncompressedsize += /*0.035*/ 0.01 * partition->partitionSizeNominal(); /* overhead for file system meta data */
```

Looking through the git history, it seems that bug has been there since 2013 :wink:
And maybe if the bug does get fixed, the "1% overhead" value would need tweaking? :man_shrugging: 

On the plus side, I guess over-estimating the needed space is much better than under-estimating the needed space :laughing:

procount, 2019-05-20:
Yeah, that's what I changed mine too.
I spotted it whilst trying to make the progress bar more accurate. 
Making good progress, but currently I have to set this tweak parameter to 5% for Raspbian_Full. Not tried any others in anger yet.