raspberrypi/noobs #544: Wifi dialog not keyboard friendly

procount, 2019-05-20:
https://github.com/raspberrypi/noobs/blob/54ac40f7c59e945d5a1389e356c4531b53ca0216/recovery/wifisettingsdialog.cpp#L189

On entering the wifi dialog, the listbox is populated with a list of found wifi SSIDs with the focus set to this list box.
If you are using cursor keys on the keyboard to move the selection down to the SSID you want, as soon as you press it once, the focus shifts to the username or password fields due to this block of code.
Not much of a problem if you are using a mouse since you can directly click on the SSID you want and will probably like the focus jumping directly to the next appropriate tab, but it's not so good when using the keyboard.

Removing this focus setting code block would make the dialog more keyboard friendly.
Or can you think of a better implementation?

duckida, 2020-12-01:
If you have an idea in mind and know how to implement it, go ahead and create a PR at [duckida/noobsplus](GitHub.com/duckida/noobsplus)

procount, 2020-12-01:
The idea I had in mind is included in the post above. It's already implemented in PINN. Perhaps you have heard of it?