raspberrypi/noobs #547: os.json file for Screenly OSE

prodigy213, 2019-07-01:
Hello,

Can someone who's familiar with Screenly or who's in contact with the maintainer for Screenly OSE please fix the os.json file located at https://releases.screenlyapp.com/NOOBS/os.json? the data inside is pointing to Raspian Lite information instead of Screenly's info. When the install is finished it loads a error message. In order for the install to work I had to use the an earlier version of the os.json file which contains the data below.

https://pastebin.com/h6zXmR7W

Thank you.

prodigy213, 2019-07-02:
> Hello,
> 
> Can someone who's familiar with Screenly or who's in contact with the maintainer for Screenly OSE please fix the os.json file located at https://releases.screenlyapp.com/NOOBS/os.json? The data inside is pointing to Raspian Lite information instead of Screenly's info. When the install is finished it loads a error message. In order for the install to work I had to revert to an earlier version. The os.json file for that install is below.
> 
> https://pastebin.com/h6zXmR7W
> 
> Thank you.

The message received. Once the installation is finsihed.

Could not expand filesystem, please try raspi-config or rc_gui.
Root partition should be last partition
<OK>

XECDesign, 2019-07-02:
@vpetersson

vpetersson, 2019-07-02:
@prescott66 My bad guys. Must have been a merge conflict gone wrong. I've updated the file, but will rebuild the image soon to accommodate for the Pi 4.

prodigy213, 2019-07-08:
@vpetersson Thanks for updating the os,json file. But Screenly still gives the same message after installation. I will create another ticket.

closed 2019-07-08 by prodigy213