raspberrypi/noobs #549: Screenly OSE fails to start after installation

prodigy213, 2019-07-09:
Hello,

I had an issue after installing Screenly OSE on a Raspberry Pi 3 Ver. 2. It gave me an message after installation. I had opened a ticket (https://github.com/raspberrypi/noobs/issues/547) hoping it would resolve the issue but it still gave the same error message. Can someone look into why after installation it gives the following message? 

Could not expand filesystem, please try raspi-config or rc_gui.
Root partition should be last partition

If logs are needed can you please tell me which one is needed?

Thank you.

lurch, 2019-07-09:
`Root partition should be last partition` is an assumption that's true for standalone images, but may not be true for NOOBS installations.
However `Could not expand filesystem, please try raspi-config or rc_gui.` isn't necessary for NOOBS installations, because NOOBS automatically creates maximum-sized partitions, so that error is probably safe to ignore. Ping @vpetersson

procount, 2019-07-09:
I agree with Lurch. The error message might be disconcerting, but it's probably benign. 
I always try to remove such resizing scripts when I convert an OS for PINN.

vpetersson, 2019-07-09:
This is strange. We're using a fork of [pi-gen](https://github.com/Screenly/pi-gen) to build our disk images. I have generated a new image, but are yet to test it. It is however strange, since it worked when I took it live.

procount, 2019-07-09:
@vpetersson - Just thinking..... Maybe you won't see the error message if you ONLY install screenly with NOOBS (since its root partition would still be the last partition). You would have to install screenly PLUS another OS that installed after Screenly for the error to be displayed?
But don't ask me how to ensure that because I still have no idea how it picks the order of OSes that it installs :) Trial and error I guess. 🤷‍♂

vpetersson, 2019-07-09:
Interesting hypothesis. I will try this when I QA the image.

vpetersson, 2019-07-10:
@procount I've pushed a new image to the repository now and I was unable to reproduce your issue. Can you confirm on a new installation?

procount, 2019-07-10:
I think it was @prodigy213 's issue, but I'll test it when I get a chance.

vpetersson, 2019-07-11:
Oh, my bad @procount.

procount, 2019-07-11:
No worries. I installed it using PINN (obviously) with Lineage (as I need to test that too) and I got no error message, but then it was installed last, so it didn't test my hypothesis. So I'm using ProjectSpaces next to force the order of installation to see if that makes a difference. 

BTW, to save me searching in the myriad of places it could be, where is your rootfs expansion script normally stored/initiated from?

vpetersson, 2019-07-11:
@procount Good find! Thanks. I will create an issue on our end to sort this out. That said, I don't think it actually breaks anything, does it?

procount, 2019-07-11:
Yep, that's the problem. Installing another OS after Screenly causes that annoying box to pop up as described. 
No it doesn't break anything, but it requires the user to cancel/accept the error message dialog box which isn't nice.
As I said, I normally remove these scripts from any OSes I convert, and it's amazing how many different ways there are to initiate a resize on boot script! Fortunately, your was an easy find and will be an easy fix.

vpetersson, 2019-07-11:
@procount Yeah we've tried to keep things as much in line with the upstream Raspbian build process as possible. I think that `cmdline` item was simply something from before we adopted pi-gen.

procount, 2019-07-11:
The script is useful for a standalone installation, just not for NOOBS/PINN.
My advice would be to modify partition_setup.sh cos it keeps it as standard as possible. Next would be to modify Pi-gen to delete it from the noobs distribution. Or finally, asks @XECDesign how he does it. :)

XECDesign, 2019-07-11:
The partition setup script pi-gen uses disables the resize script.

https://github.com/RPi-Distro/pi-gen/blob/master/export-noobs/00-release/files/partition_setup.sh#L39

vpetersson, 2019-07-18:
@XECDesign Yep, my bad. Our script runs after that I think, and presumably it reverts the setting.

vpetersson, 2019-07-18:
Is anyone still having issues? We have not heard any issues reported on this since. Pi 4 support is coming, but still in the works.

XECDesign, 2019-07-18:
Thanks, Viktor. I haven't seen any complaints either, so it's probably good to close.

closed 2019-07-18 by XECDesign

prodigy213, 2019-07-18:
I just did an install using PINN v3.3 Built Jun 22 2019 on a Raspberry Pi 3 Ver. 2 Model B. It's still giving me the same message when installing Screenly with multiple OSes.

Screenly will ONLY work if it's the last OS installed or if it's a standalone.

vpetersson, 2019-07-18:
@prodigy213 I've only verified it using NOOBS tbh. I did however verify it with NOOBS with multiple installations successfully.

prodigy213, 2019-07-18:
@vpetersson Awesome! Thank you.

lurch, 2019-07-18:
@prodigy213 I've never used Screenly, but based on earlier messages in this issue, do you really mean "will ONLY work if..." or "will only not display harmless error messages if..." ?

procount, 2019-07-18:
@lurch - I think that is a better summary statement as the error message is benign.

@vpetersson - the issue will remain until you remove the partition resize script. However, it will only present itself as an unwelcome error dialog box IFF:
1) Screenly is installed with multiple OSes (whether using NOOBS or PINN)
2) Screenly is NOT the last OS to be installed on the SD card.
You did tests with the first condition, but did you ensure the second condition was also met? (You can check the installed order by examining /settings/installed_os.json IIRC). It is not easy to force the order of OS installation with NOOBS and in my tests Screenly was usually installed last, thus not presenting the error. But I can force the order with PINN using its ProjectSpaces feature.

I don't see it as a big deal since it's just a benign warning message that needs to be acknowledged and doesn't otherwise affect the installation. It's more of an occasional annoyance. But I'm sure it will be reported again if not fixed. It sounds like an easy change to revert back to pi-gen's functionality that could be done when your Pi4 version is released.

prodigy213, 2019-07-20:
@lurch At the moment I'm using PINN v3.3 Built Jun 22 2019 on a Raspberry Pi 3 Ver. 2 Model B. When Screenly is the last OS to be installed or if it's a standalone, it works properly.

If you load up a 32GB or 64GB SDCard with multiple OSes (lets go with Raspbian, Lakka and Recalbox) including Screenly OSE. If Screenly is installed before the other OSes it will display. "Could not expand filesystem, please try raspi-config or rc_gui. Root partition should be last partition". At that point the only option is to press "ok". With that action, the system will reboot and go back to the OS selection menu. If you go back into Screenly the same thing will happen.

I have not had a chance to check @vpetersson statement using NOOBS. (FYI I also installed Screenly using NOOBS the first go around). I switched to PINN because of the reinstall and OS swap features (including the awesome list of OSes). I will check in a few days because I'm currently busy with other projects.

procount, 2019-07-25:
@prodigy213 - This issue should now be fixed for PINN users as I have overridden the partition_setup.sh script for screenly. I just tested it with an install of Raspbian_LIte which was installed 2nd, and the error message did not occur. NOOBS users may still have a problem.

@vpetersson - I'm not sure if you've updated your partition_setup.sh script because I didn't take a copy previously. However, looking at it, it should now work **EXCEPT** the sed substition includes the word "quiet" which is no longer relevant since your cmdline.txt file has changed, and hence the substitution fails.

May I suggest you replace your script with my adjusted version from https://github.com/procount/pinn-os/tree/master/partitions/Screenly/partition_setup.sh. 

It also includes some additional features to make it work better with PINN, yet still remain NOOBS compatible. See lines 2, 20, 36. The SED modification is in line 34. 

My sed expressions on lines 17&18 are a bit different to yours. I don't think they make much difference in this case, so include or exclude them as you wish.

prodigy213, 2019-07-29:
@procount YOU'RE THEE MAN! THANKS SO MUCH!!! I just did a reinstall on PINN and it works perfectly. Issue resolved. :-)

@lurch I was trying to perform the test on a 8GB then a 16GB microSD card using NOOBS but Screenly kept installing last. It's almost impossible to get OSes installed in the order you'd like. So I wasn't able to confirm. If @vpetersson implements @procount fix everything should be okay, since the installation worked on PINN.

vpetersson, 2019-08-01:
@rusko124 - Can you confirm that this was resolved in https://github.com/Screenly/pi-gen/pull/15 ?