raspberrypi/noobs #550: Allow NOOBs to use Bluetooth Keyboard/Mouse and configure OS

DonBower, 2019-08-17:
#257 closed over a year ago, but I think we still have a need for this.
I would like to pre-configure NOOBs with a bluetooth .conf file, similar to the wpa_supplicant.conf file.
perhaps bt_supplicant.conf:
```
Device 20:73:AB:75:73:C8 bluetooth mouse4.0
Device EC:23:45:67:EE:B7 Bluetooth Keyboard
```
so that when NOOBs is booted, not only would we be able to "Hold Shift for Recovery", but we could select other configuration items, like which OS, etc during initial install

lurch, 2019-08-18:
@procount Is this something that PINN supports?

XECDesign, 2019-08-19:
Nothing has changed since the issue was closed last time. There are no immediate plans to add new features to NOOBS in the near future.

procount, 2019-08-19:
No, PINN does not support Bluetooth.

DonBower, 2019-08-19:
Perhaps this position should be revisited, with the proliferation of Bluetooth Keyboards/Mice....

candoattotude, 2020-07-03:
Agreed @DonBower. I have to connect an extra keyboard just for the noobs OS selection. Bluetooth compatibility would be much appreciated.