raspberrypi/noobs #551: boot {part} not working on Pi4 (Noobs 3.2)

matthuisman, 2019-08-27:
NOOBs Lite 3.2
https://downloads.raspberrypi.org/NOOBS_lite_latest

Current Behaviour
NOOBS ignores reboot {part} and boots to normal NOOBS screen

Expected Behaviour
reboot {part} reboots into part

Steps to reproduce
- Format SD card and copy over NOOBs 3.2 files
- Boot NOOBs and install Raspbian Lite
- Boot Raspbian Lite
- Login (pi / raspberry)
- Run command: **sudo reboot 6**
- Observe NOOBS reboots into colour screen then NOOBS then raspbian
- Should skip NOOBS completely and boot straight into Rasbpian

Notes
autoboot.txt still works correctly

XECDesign, 2019-08-27:
This isn't a NOOBS issue, but a systemd issue. It may be fixed in Debian's next point release, but I'm not sure.

I've reported it here a while back: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=928659

matthuisman, 2019-08-27:
cc @procount (also same with latest PINN)

If I remember correctly, the reboot {part} stuff is actually inside the Raspberry Pi firmware?

XECDesign, 2019-08-28:
Alright, might be another issue then. I'll check.

matthuisman, 2019-08-28:
ok, looks like it was merged into systemd here:
https://github.com/systemd/systemd/pull/11602

I suspect it may never find it's way into buster?
So probably best I start looking for a work-around.

My main requirement is it working in LibreELEC.
For my noobs companion add-on to easily boot into other installed OS's.

I do have a a bit of reboot code I use on systems that didn't support reboot {params}
https://github.com/matthuisman/plugin.program.noobs.companion/blob/master/resources/files/part_reboot.c

So maybe I can use that for now on these.
However, not sure how to do a tidy shutdown before the syscall.

UPDATE

I've done same tests in LibreELEC across both boards and the fault is only on RPI4.

LibreELEC 9.1.002 Pi3B+ (systemd version 242)
reboot 6  - works

LibreELEC 9.1.002 Pi4 (systemd version 242)
reboot 6  - doesn't work

Same results with latest master builds from here:
http://milhouse.libreelec.tv/builds/master/

I also tried my reboot_part code which bypasses systemctl and uses the syscall with LINUX_REBOOT_CMD_RESTART2 and that only works on Pi3B+

I'll update my first post to use LibreELEC for confirming fault as it removes systemctl from equation.

Leads me to believe that it is broken somewhere in Raspberry Pi 4 firmware?

XECDesign, 2019-08-28:
The fix to the issue I thought this was was introduced in `241-4`, so yes, it's definitely a part of Raspbian now.

Sorry, I completely forgot about this, but you can't reboot into different partitions when the uSD interface is running on 1.8V (a higher speed mode). You need to boot with `sdhci.debug_quirks2=4` in cmdline.txt for it to work, but it will sacrifice SD speed (I don't know how much).

AIUI, the issue is that once the card is in UHS (1.8V) mode, we can't get out of it without resetting power, losing the content of the reset register (which contains the reboot partition number). Due to reasons I don't quite understand, we can't continue in UHS mode and must switch to 3.3V to properly initialise the card.

lurch, 2019-08-28:
Background info about the bewildering array of different SD card modes https://en.wikipedia.org/wiki/SD_card#Ultra_High_Speed_(UHS)

ping @procount as he'll probably find this thread useful for PINN :wink:

matthuisman, 2019-08-28:
Interesting.
So the Pi3B+ doesn't use UHS?

matthuisman, 2019-08-28:
> Correct, prior to the pi 4, we didn't do UHS switching.

So the cmdline quirk just goes in the noobs cmdline? Or will each OS need
it?
If just in noobs, then the slowdown would only be in noobs itself?

matthuisman, 2019-08-28:
> It needs to into cmdline.txt for each partition you'd like to reboot
> from/through. If at any point it goes into UHS mode, you'll lose the
> ability to reboot into other partitions.
>
> By default, we only add it to NOOBS itself, which is why rebooting into
> different partitions works from NOOBS, but not after booting into an OS
> without the quirk set.

Oh thats a shame :(
Noobs would probably enjoy UHS speeds when installing OS's etc.

Do you think a fix is likely?
Eg. Storing reboot param elsewhere that can survive the power loss.

Or, can you think of any work around?
I was thinking writing to autoboot but then no way to remove it once booted.

Maybe a one shot file could be added?
Works same as autoboot, but only once (eg. Removed after used)

Or maybe a software solution in noobs where it could be told to autoboot an
os (as quickly as possible)

Noobs reads file then does reboot to partition. No recovery etc. Won't be
as quick as reboot param but still something.

XECDesign, 2019-08-28:
Correct, prior to the pi 4, we didn't do UHS switching.

XECDesign, 2019-08-28:
It needs to into cmdline.txt for each partition you'd like to reboot from/through. If at any point it goes into UHS mode, you'll lose the ability to reboot into other partitions.

By default, we only add it to NOOBS itself, which is why rebooting into different partitions works from NOOBS, but not after booting into an OS without the quirk set.

XECDesign, 2019-08-28:
The only two places that would survive power loss (that I know of) are the EEPROM and the SD Card.

It can't be the SD card because we can't write to it when it's booted straight into UHS mode. At least not in any way that we think is safe.

It can't be the EEPROM because it will complicate the bootloader and waste write cycles.

procount, 2019-08-29:
> ping @procount as he'll probably find this thread useful for PINN 😉

Now that I am reminded, I remember XECDesign mentioning this issue to me when I was first adapting for the Pi4.

> Oh thats a shame :(
> Noobs would probably enjoy UHS speeds when installing OS's etc.

@matthuisman - Some idea of the speed difference would be interesting....

> Or maybe a software solution in noobs where it could be told to autoboot an
> os (as quickly as possible)

From what I understand from @XECDesign 's remarks, no matter what software solution you put in place to circumvent this, you still need to pull the power to allow the SD card to switch out of UHS mode. i.e even modifying autoboot.txt (or similar) won't work unless you pull the power to cause the reboot - is that correct?

XECDesign, 2019-08-29:
Not pull the power in the sense that you need to unplug it. Just that when the bootloader starts and detects that it's in UHS mode, it will reset itself in a way that clears the reset register.

If you wanted to, you could drop a file somewhere and have PINN look at it and then delete it. Of course that would be much slower, since it would have to go through PINN. You could drop autoboot.txt and then delete it on next boot, but that would be up to each OS you boot into. Not sure that I can think of many other alternatives.

lurch, 2019-08-29:
> Not pull the power in the sense that you need to unplug it. Just that when the bootloader starts and detects that it's in UHS mode, it will reset itself in a way that clears the reset register.

Ahhh, I was wondering if that was the case, because otherwise a 'sudo reboot' wouldn't work from a regular image-install of Raspbian either :wink: 

> You could drop autoboot.txt and then delete it on next boot, but that would be up to each OS you boot into.

Yeah, that's what I was thinking too, but you'd probably also want some other "marker file" indicating that this is only a "temporary" autoboot.txt which should be auto-deleted, to prevent "normal" usages of autoboot.txt being deleted automatically. (IYSWIM!)

procount, 2019-08-29:
Oh OK, so whether it's a cold (pull the plug) or warm (software reset) reboot, it's the startup firmware that will reset the SD card out of UHS mode by removing power from it (or similar) that has the side effect of clearing the reset register.

So, in PINN, I already have the "sticky boot" flag, which I think may do most of the required behaviour already. But obviously it's not as quick as autoboot.txt.

procount, 2019-08-29:
> It can't be the SD card because we can't write to it when it's booted straight into UHS mode. At least not in any way that we think is safe.

Can you expand on this, because it doesn't make much sense to me?

> we can't write to it when it's booted straight into UHS mode.

Obviously, the SD card can be written to by Linux when in UHS mode, so is this just a GPU firmware issue, or a timing issue, or .... ?

lurch, 2019-08-29:
I suspect the "problems" are all in the SD card initialisation sequence - the initialisation sequence (which will need to run every time the Pi powers up) probably gets confused (on the Pi and/or on the SD card) if the card is already in UHS mode? Just guessing!

XECDesign, 2019-08-29:
Linux doesn't start with the card in 1v8 mode, it's always 3v3 mode right now.

A very rough explanation is that when you start talking to a card, you start in 3v3 mode and you ask what modes it supports. Then you can switch to 1v8 and go from there. This is done when linux detects the card is there. Due to a hardware limitation on the pi, once you're in 3v3 mode you can't get out of it without letting the SoC reset itself.

So, if the bootloader starts and detects that it's in 1v8 mode already, it's not safe to just assume the card supports UHS, skip initialization and just go from there. If you ignore that and proceed anyway, and it works and there are no issues, then you run into Linux having to do the same thing when it starts. 

I don't know why it's not safe, but that's what I'm told by people who understand the issue much better than I do. Also, you'd be asking the one guy who's working on the bootloader to take time away from working on network and USB boot to focus on a potentially unsafe niche use case that's just "nice to have".

procount, 2019-08-29:
OK, thanks. That makes more sense now. Probably the SD card controller doesn't have a separate controllable power line, so the only way to reset it is to reset the SOC, hence the issue. 
I shall have to see if my "sticky boot" flag is a possible (but slow) way around this. Maybe I'll have to make setting it easier from another OS, but I'm afraid it's low priority as I have some other things to sort out first.

XECDesign, 2019-08-29:
> Probably the SD card controller doesn't have a separate controllable power line, so the only way to reset it is to reset the SOC, hence the issue.

AFAIK, that's spot on.

matthuisman, 2019-08-29:
@procount

I had suggested something like that.
A one-shot autoboot file.
Sticky boot is similar, but this would be faster with no recovery at all.
To get close to possible as reboot {param}

However, the file method is still tricky for OS's to use.
It would need to mount the pinn partition, and create the file.
A lot more involved than reboot {param}

I think it's just going to have to be 
"Doesn't work on RPI4, you need to manually choose OS each time"

procount, 2019-08-29:
You could implement a one-shot autoboot file using PINN"s pinn_init.sh script. I think that's the fastest you can do it without modifying the firmware,

matthuisman, 2019-08-29:
so something like this I guess

```
FILE=one_shot.txt
if test -f "$FILE"; then
    PART=$(cat $FILE)
    rm $FILE
    reboot $PART
fi
```

procount, 2019-08-29:
Yep. That sort of thing, except you have to use `rebootp $part` in PINN.

matthuisman, 2019-08-29:
Thanks. 
Updated the code for others reference.
Closing this issue now as can't really be fixed.

closed 2019-08-29 by matthuisman

soundstorm, 2020-01-10:
@matthuisman would you mind to add the cmdline.txt instructions to your instructions on your website tutorials? Spent the half night to figure out what was wrong 🙈