raspberrypi/noobs #552: Is NOOBS repo not updated anymore?

carlonluca, 2019-08-28:
I downloaded NOOBS from the raspberry pi official website. I see that NOOBS version is v3.1, release date is 2019-06-10 and commit is a867659d7e5af44db87969e4ccd3fea0370a9cd1. I see the latest commit is Apr 15, 2019. Is it now versioned somewhere else? Thanks!

XECDesign, 2019-08-28:
At the time 3.2 was released, it was built from linux and firmware repos which are no publicly accessible. It's a little hard to put the source up, but if it's requested, I can do so and attach tarballs for kernel source and firmware. It would just be a little bit for a pain.

carlonluca, 2019-08-28:
Hello, thanks, no need to waste time for the moment. Just wanted to know if this is still the repo and if I should expect new updates to arrive here when ready.
Thanks!