raspberrypi/noobs #553: Raspberrypi 4B cannot load noobs

zhjgit, 2019-09-02:
I installed the tutorial on the official website and formatted my SD card, because the SD card exceeded 32G. I converted the file system into FAT32, and then decompressed the new version of noobs Lite into my SD card. I tried many times, including rewriting and formatting the full coverage of the card, and could not load noobs.

zhjgit, 2019-09-02:
Now I want to try using noobs lite image file and burn it to the SD card. I don't know if it's possible, if it's possible, to help create noobs image files.

XECDesign, 2019-09-02:
For general support, please use the forum.
https://www.raspberrypi.org/forums/index.php

https://www.raspberrypi.org/documentation/installation/noobs.md
https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md

zhjgit, 2019-09-02:
At the beginning, the green light went on a few times, and then the red light went on all the time. And it's always been this picture.
[image]

JamesH65, 2019-09-02:
Again, please use the forum for technical support of this sort. Links provided above.

closed 2019-09-02 by JamesH65

procount, 2019-09-02:
Did you plug your HDMI cable into the HDMI0 socket - the one next to the USB-C connector?

zhjgit, 2019-09-02:
@procount I succeeded with the HDMI interface near the power supply. Why? Are there any differences between these two HDMI interfaces?

procount, 2019-09-02:
Noobs only displays on HDMI0.