raspberrypi/noobs #554: Lakka RasPi issue

bdspvl, 2019-09-05:
Hello all new user on GitHub! Have an issue that started yesterday, Doing a now boot SD card for Pi and found out with Noobs, the Lakka RP2 DIY Retro emulation console has error and no download. Found out the download area is no longer there. Does anybody have a copy of the files and if so tell how to do a manual install without NOOBS? Or better still notify whoever to rebuild the site for downloading?

lurch, 2019-09-05:
Appears to be the same as https://github.com/libretro/Lakka-LibreELEC/issues/769 (i.e. _none_ of the Lakka downloads are working?)

bdspvl, 2019-09-06:
> Appears to be the same as libretro/Lakka-LibreELEC#769
> <https://github.com/libretro/Lakka-LibreELEC/issues/769> (i.e. *none* of
> the Lakka downloads are working?)

I don't know for sure. It was working over the weekend. Was busy moving up
to a 30 gig sd card when I noticed it. Thought something was wrong with the
SD card at first. Reformatted ok. Today looked at link and nothing is
there! Are there any backup locations on the net that anybody knows of?
Makes me think I should of made an image copy.. darn! Put a note out to see
if anybody has a new install copy!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

XECDesign, 2019-09-06:
@kivutar

XECDesign, 2019-09-09:
Lakka removed from os_list.

kivutar, 2019-09-10:
@natinusala

natinusala, 2019-09-10:
We had an internal issue that caused all our images to be removed from the build server - we are currently working on releasing Lakka 2.3. I will let you know when the files are back online, sorry for the trouble

bdspvl, 2019-09-14:
Sounds good!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

natinusala, 2019-09-18:
Okay, the regular images are back online.

We have three different NOOBS archives: one for RPi, one for RPi2 and 3 and one for RPi4

Should I upload all three? How do you want to proceed?

XECDesign, 2019-09-18:
All three would be alright.

natinusala, 2019-09-18:
Sorry I'm unfamiliar with how NOOBS gets the files, I have to extract all the files of the archive to a webserver right? Or upload the whole archive as it is?

XECDesign, 2019-09-18:
You can use libreelec as a template. Look at their section here http://downloads.raspberrypi.org/os_list_v3.json

```
        {
            "description": "LibreELEC is a fast and user-friendly Kodi Entertainment Center distribution.",
            "icon": "http://releases.libreelec.tv/noobs/LibreELEC_RPi/LibreELEC_RPi.png",
            "marketing_info": "http://releases.libreelec.tv/noobs/LibreELEC_RPi/marketing.tar",
            "nominal_size": 1024,
            "os_info": "http://releases.libreelec.tv/noobs/LibreELEC_RPi/os.json",
            "os_name": "LibreELEC_RPi",
            "partition_setup": "http://releases.libreelec.tv/noobs/LibreELEC_RPi/partition_setup.sh",
            "partitions_info": "http://releases.libreelec.tv/noobs/LibreELEC_RPi/partitions.json",
            "release_date": "2019-05-04",
            "supported_hex_revisions": "2,3,4,5,6,7,8,9,d,e,f,10,11,12,14,19,0092",
            "supported_models": [
                "Pi Model",
                "Pi Compute Module Rev",
                "Pi Zero"
            ],
            "tarballs": [
                "http://releases.libreelec.tv/noobs/LibreELEC_RPi/System.tar.xz",
                "http://releases.libreelec.tv/noobs/LibreELEC_RPi/Storage.tar.xz"
            ]
        },
```

We just need to be able to reach each of the files os_list references. So I'd expect to see a file structure that looks like this http://releases.libreelec.tv/noobs/

natinusala, 2019-09-19:
Okay thanks for the info

Do the three versions need to have a different `os_name`, or can the three of them be `Lakka`?

lurch, 2019-09-19:
I think it's probably best to give them a different name, in case somebody e.g. wants to use the same SD card  on both a Pi1 and a Pi3? (and I've not checked the code, so I dunno what NOOBS would actually do if it did find multiple OSes with the same os_name?)

natinusala, 2019-09-19:
Okay then I will upload the images with the 2.3.1 release in a few days, sorry for all the fuss

I will let you know here when all the files are published and ready, I'll also give you the os_list.json entry for all variants if needed

natinusala, 2019-09-20:
Here you go

```
{
    "description": "The DIY retro emulation console",
    "icon": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/Lakka_RPi4.png",
    "marketing_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/marketing.tar",
    "nominal_size": 1024,
    "os_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/os.json",
    "os_name": "Lakka_RPi4",
    "partition_setup": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/partition_setup.sh",
    "partitions_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi4/partitions.json",
    "release_date": "2019-09-20",
    "supported_hex_revisions": "",
    "supported_models": [
        "Pi 4"
    ],
    "tarballs": [
        "http://le.builds.lakka.tv/noobs/Lakka_RPi4/System.tar.xz",
        "http://le.builds.lakka.tv/noobs/Lakka_RPi4/Storage.tar.xz"
    ]
},
{
    "description": "The DIY retro emulation console",
    "icon": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/Lakka_RPi2.png",
    "marketing_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/marketing.tar",
    "nominal_size": 1024,
    "os_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/os.json",
    "os_name": "Lakka_RPi2",
    "partition_setup": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/partition_setup.sh",
    "partitions_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi2/partitions.json",
    "release_date": "2019-09-20",
    "supported_hex_revisions": "1040,1041,2082",
    "supported_models": [
        "Pi 2",
        "Pi 3"
    ],
    "tarballs": [
        "http://le.builds.lakka.tv/noobs/Lakka_RPi2/System.tar.xz",
        "http://le.builds.lakka.tv/noobs/Lakka_RPi2/Storage.tar.xz"
    ]
},
{
    "description": "The DIY retro emulation console",
    "icon": "http://le.builds.lakka.tv/noobs/Lakka_RPi/Lakka_RPi.png",
    "marketing_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi/marketing.tar",
    "nominal_size": 1024,
    "os_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi/os.json",
    "os_name": "Lakka_RPi",
    "partition_setup": "http://le.builds.lakka.tv/noobs/Lakka_RPi/partition_setup.sh",
    "partitions_info": "http://le.builds.lakka.tv/noobs/Lakka_RPi/partitions.json",
    "release_date": "2019-09-20",
    "supported_hex_revisions": "2,3,4,5,6,7,8,9,d,e,f,10,11,12,14,19,0092,0093"
    "supported_models": [
       "Pi Model",
       "Pi Compute Module",
       "Pi Zero"
    ],
    "tarballs": [
        "http://le.builds.lakka.tv/noobs/Lakka_RPi/System.tar.xz",
        "http://le.builds.lakka.tv/noobs/Lakka_RPi/Storage.tar.xz"
    ]
}
```

natinusala, 2019-09-20:
I just left `nominal_size` at `1024` because I don't know what it refers to

lurch, 2019-09-21:
IIRC it's the sum of the `partition_size_nominal` values in `partitions.json` ?  (which the NOOBS GUI uses to decide if there's enough free space on the SD card to install the OS)

natinusala, 2019-09-21:
Then 1024 is correct

procount, 2019-09-23:
@natinusala 
Please could you add the following to each of your OS entries for better support in PINN:
```
        "url": "http://www.lakka.tv/",
        "group": "Games",
        "supports_backup" : true,
        "download_size": XXXXX,
```
Where XXXXX is the sum of the filesizes (in bytes) of system.tar.xz and storage.tar.xz for each OS respectively.

natinusala, 2019-09-25:
@procount done!

XECDesign, 2019-09-26:
Should be in os_list_v3.json now

natinusala, 2019-09-26:
Excellent, thank you very much

lurch, 2019-09-26:
@bdspvl Would you like to test this, and close this issue if it's fixed?

XECDesign, 2019-09-26:
Trying to test it, but the server seems a little on the slow.

procount, 2019-09-26:
@natinusala - Thanks for adding the extra fields for PINN.
Could they be added to the entries in `os_list_v3.json` too?
(In fact, the download_size entry only makes sense in `os_list_v3.json` and can be deleted from `os.json`, but it's not essential.)
Maybe @XECDesign could assist?

XECDesign, 2019-09-26:
Manually updating download_size in os_list_v3.json each time isn't something I'm keen on.

XECDesign, 2019-09-26:
> download_size entry only makes sense in os_list_v3.json

Does that imply the other entries make sense in os.json? What's the difference in how they're handled in the two files?

lurch, 2019-09-26:
@procount Just a suggestion, but I wonder if you could modify PINN to work out the (total) size of the download tarballs using http HEAD requests?

procount, 2019-09-26:
> Does that imply the other entries make sense in os.json?

Yes.
> What's the difference in how they're handled in the two files?

 os_list_v3.json contains just enough meta-data for "remote" OSes on a server to be displayed in NOOBS/PINN and downloaded. For these OSes, os.json is not actually used.
When the OS is stored locally on the SD card (like in NOOBS-FULL) or on the USB stick, there is no os_list_v3.json, so os.json is used instead. (Partitions.json is used for both local and remote OSes.)
I don't know why it is like that, it's  before my time!
So the extra fields need to be in both files.

"download_size" is only used for progress indication, and does not have to be accurate, although it does help. So you don't need to update it every time. In any case, I can override it if it gets too far out of step with each update.

But on a new release of an OS, you have to update the OS entries in os_list_v3.json anyway (release date, version, description maybe), so what is the difference if the maintainer can update this field? You just need to cut n' paste the whole OS entry into os_list_v3.json.

@Lurch, in fact PINN already does calculate the download size for each OS, but only if the `download_size` field is not present. It takes a relatively long time to calculate it for 60-ish OSes, so it slows the startup time down. This was a way of caching the data to speed it up. Likewise I cache the OS icons to speed up the startup.

XECDesign, 2019-09-26:
Sounds good, I'll copy the entries. Thanks, Kevin.

procount, 2019-09-26:
> Maybe @XECDesign could assist?

I just meant "on this occasion" (in case you took it the wrong way). Thereafter it would be ideal if the maintainers could update this field themselves on each update.

natinusala, 2019-09-26:
> in fact PINN already does calculate the download size for each OS, but only if the download_size field is not present. It takes a relatively long time to calculate it for 60-ish OSes, so it slows the startup time down.

But why do you need to know the size of every OS? Can't you do the size check only when the user wants to install the OS?

XECDesign, 2019-09-26:
Added

procount, 2019-09-26:
> Can't you do the size check only when the user wants to install the OS?

Probably because it fitted with the current workflow of gathering all information upfront before the user selects any OS. But it's an interesting suggestion so I may consider it for a future version.

@XECDesign  - thanks!

lurch, 2019-09-26:
> It takes a relatively long time to calculate it for 60-ish OSes, so it slows the startup time down.

I'd forgotten how many OSes you have in PINN! :bowing_man:

procount, 2019-09-26:
Well, it depends on the model of RPi as to how many are available, but I still have to read them all in. 

According to http://pinn.mjh.nz/ there are 62 OSes available for the 3B, 20 for the 4B and 47 for most others, all spread over 4 separate source repositories. Which explains why it sometimes takes me a while to update them to the latest version!

procount, 2019-09-26:
@natinusala - I don't see any "supports_usb" options, so I'm assuming these latest lakka versions are ok to install to usb (either fully or just the rootfs)?

I ask because some PINN users using the previous Lakka versions had found it was possible to install them to USB, but were blocked from doing so due to these options previously being set to false.

natinusala, 2019-09-26:
I'm not sure, I think the RPi2/3 version can but not the RPi4 (because the bootloader doesn't support it?)

procount, 2019-09-26:
Of course, pi4 excepted for the time being.