raspberrypi/noobs #555: Kali img with NOOBS

Nexobeta28, 2019-09-29:
Hi, first of all, sorry for the dumb question, I got my raspi 3b+ 3 days ago and Im still learning. I was trying to use my kali img with NOOBS, so I could use raspbian, recalbox and kali. In older posts, the people was saying something about convert the img into a tar.xz, but the os's installed with NOOBS are in a folder, not compressed. I'm a bit confused and I can't find any information about this. If anyone can post a link to a kali os prepared for NOOBS would be great.

Sorry for my poor english :P

procount, 2019-09-30:
There are some Kali images already prepared that you can install with [PINN](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574). They are a little old and I need to update them to the latest versions and to support the Pi4.

lurch, 2019-10-02:
> os's installed with NOOBS are in a folder, not compressed.

NOOBS supports installing OSes directly from the recovery partition on the SD card (which I guess is the folder you're talking about), or it can also install them by downloading them directly from the internet.

XECDesign, 2019-10-02:
For general support, please use the forum.

https://www.raspberrypi.org/forums/index.php

lurch, 2020-03-24:
@Nexobeta28 Did you manage to get Kali installed using PINN? Can this issue be closed?

alesbe, 2020-09-29:
Sorry, I was looking at my contributions and saw this. I think that I managed to Install kali finding an old ISO made for RPi.

closed 2020-09-29 by alesbe