raspberrypi/noobs #557: custom OS with NOOBS lite

eroom1966, 2019-10-11:
Hi
I have my own custom (configured) OS which I have been successfully installing as part of a full NOOBS installation, I want to be able to do the same from a NOOBS-lite style installation
in other words
I will host the boot.tar.xz and root.tar.xz on my own web connected file server
I simply want to auto install after download, rather than including the .xz files on the SD card

I cannot work out how to do this, various documentation points at .json files, but it appears that in a NOOBS-lite installation, there are NO .json files, so it would appear it is hard coded somewhere

any advice welcome
Thx
Lee

lurch, 2019-10-11:
You may be better off using [PINN](https://github.com/procount/pinn), as it has more support for these types of customisations?
ping @procount

eroom1966, 2019-10-11:
@lurch 
That may be an option, but would mean reworking a mass of infrastructure to do that.
so if I can simply resolve this by accessing any 'low hanging fruit' that will be a far better approach
I have a suspicion that Noobs lite DOES look for a local .json file in addition to pulling the web based file, in which case - it is simply a case of matching the intended method it uses
unfortunately, I cannot find any documentation to prove this, and may result in trawling through code to reverse engineer the intent
Thx
Lee

XECDesign, 2019-10-11:
You can drop os_list_v3.json onto the card or specify a URL in recovery.cmdline

https://github.com/raspberrypi/noobs/blob/master/recovery/mainwindow.cpp#L165

lurch, 2019-10-11:
@XECDesign Cool, I can never remember which features are in NOOBS and which are only in PINN :wink:

eroom1966, 2019-10-11:
@XECDesign 
Excellent!
This is what I was hopeful for, can you please confirm the exact location it must appear in the SD card hierarchy ?
I presume "/mnt/os_list_v3.json" is mounted from some location on the SD ?
likely candidates I guess are 
/os_list_v3.json
or /os/list_v3.json
or /defaults/list_v3.json

Thx in advance
Lee

XECDesign, 2019-10-11:
/

lurch, 2019-10-11:
I think in this particular snippet of code, `/mnt` is the first FAT partition on the SD card, i.e. the same one that contains `recovery.rfs`, etc.

procount, 2019-10-11:
Yes.
`os_list_v3.json` should be on `/dev/mmcblk0p1`.
If you are executing NOOBS-lite and drop into the recovery shell with ctrl-alt-f2, then /dev/mmcblk0p1 is mounted read-only on /mnt.
In this case, you need to do: `mount -o remount,rw /mnt` to make it writeable.
But if you are copying the file onto the SD card when in another PC, then it's the root of the first partition, where ever you mount it. (Where recovery.rfs is stored, as Lurch noted).

procount, 2019-10-11:
For the command line option to load `os_list_v3.json` from your "web connected file server", you can add `repo=http://blahblah.com/os_list_v3.json` to recovery.cmdline.

procount, 2019-10-11:
> That may be an option, but would mean reworking a mass of infrastructure to do that.

In the majority of cases, PINN is just a drop-in replacement for NOOBS, so I'm not sure what "mass of infrastructure" changes you would need to do...

eroom1966, 2019-10-11:
> 
> 
> > That may be an option, but would mean reworking a mass of infrastructure to do that.
> 
> In the majority of cases, PINN is just a drop-in replacement for NOOBS, so I'm not sure what "mass of infrastructure" changes you would need to do...

OK, I will give that a go
Thx
Lee

procount, 2019-10-14:
Glad it worked for you. 
BTW - I wasn't advocating that you change from NOOBS to PINN. That's up to you and depends on whether you want the extra features or not.
I was just pointing out that PINN is a fork of NOOBS and so shares the same infrastructure.
Dropping your os_list_v3.json will work just the same on PINN. But as always - do it on a fresh or cloned SD card first, just in case!

XECDesign, 2019-10-14:
ok to close?

eroom1966, 2019-10-14:
@XECDesign 
yes - many thanks

eroom1966, 2019-10-15:
> Glad it worked for you.
> BTW - I wasn't advocating that you change from NOOBS to PINN. That's up to you and depends on whether you want the extra features or not.
> I was just pointing out that PINN is [...]

OK, can you point me at something to read to understand the diffs between PINN and NOOBS ?
If PINN is a better solution, I am happy to spend time porting across

procount, 2019-10-15:
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574
https://github.com/procount/pinn
https://github.com/procount/pinn/blob/master/README.md
https://github.com/procount/pinn/blob/master/changes.md <- started out as the differences between NOOBS abd PINN but deteriorated into just a changelog
https://github.com/procount/pinn/blob/master/README_pinn.md

eroom1966, 2019-11-02:
I have a subsequent question 
I cannot seem to get this flow to work with the additional command line options 

silentinstall
Or
silentinstall forcetrigger

Any clues why ?

lurch, 2019-11-03:
IIRC `silentinstall` only works if one OS is available, so I guess you need to remove "any other OSes" to ensure that your custom OS is the only one available?

eroom1966, 2019-11-03:
@lurch 
Thanks, I only have one OS available 
I think having read some of the other issues, that this is not supported, but may be supported in PINN
So heading over there ...

XECDesign, 2019-11-04:
I haven't tried the option in a long time, but I'm not aware of any reason for it to have stopped working.

Closing. Not because I don't think it's possibly any issue, but because it's not related to the original issue.

closed 2019-11-04 by XECDesign