raspberrypi/noobs #558: Error downloading or extracting tarball

paul-clark-eprint, 2019-10-25:
I have been using the instructions on this site "How to create a custom OS version" to create to files boot.tar.xz and root.tar.xz to create a custom version of Noobs. Since the Raspbian Buster release, I get an error "Error downloading or extracting tarball" with lots of files listed starting with:
.dev/hidraw1: Can't create 'dev/hidraw1'
.dev/fb0: Can't create 'dev/fb0'
... there is a long list of files with the same error.
I'm guessing that there has been a change in the way the tar files should be created. Can you help me out please?
[image]

Paul

XECDesign, 2019-10-25:
There haven't been any changes in how the tarballs are created.

If I was to guess, it looks like the tarballs were created from a running system. Those dev nodes look likes ones that are normally created at runtime.

maxnet, 2019-10-25:
You did use the --one-file-system option when creating the tar?
As even if you tar on a running system, that should not include /dev because it is on other (devtmpfs) mount point.

procount, 2019-10-25:
Did you use the exact options to bsdtar mentioned? As XECDesign mentioned, the dev nodes shouldn't appear in the tarballs.
I detailed a more comprehensive tutorial on creating custom OS's on my wiki at https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN
(maxnet - you just beat me to it!)

paul-clark-eprint, 2019-10-25:
Thanks for the replies. Sorry, I had made an error while making the boot tar file. All good now

closed 2019-10-25 by paul-clark-eprint