raspberrypi/noobs #567: Rainbow splash image is shown despite disabling in /boot/config.txt

slrslr, 2020-01-11:
Hello,

i used NOOBS_v3_2_1 on RPi4B and installed full Raspbian 10.
I tried to set disable_splash variable in /boot/config.txt as per the instructions here:
https://www.raspberrypi.org/documentation/configuration/config-txt/boot.md

But it continue to appear.

boot mount points:

$ mount|grep boot

> /dev/mmcblk0p6 on /boot
> /dev/sda6 on /media/pi/boot

Finding all config.txt files in / and grepping splash out of its contents:

$ for file in $(find / -type f -name config.txt 2>/dev/null);do echo $file && grep splash $file;done 

> /boot/config.txt
> disable_splash=1
> /media/pi/boot/config.txt
> disable_splash=1

procount, 2020-01-11:
You are probably seeing the splash screen from noobs.
Create a config.txt file on your noobs recovery partition (/dev/mmcblk0p1) and add disable_splash=1 there.

slrslr, 2020-01-11:
that worked,
mkdir /media/recovery;mount /dev/mmcblk0p1 /media/recovery
cp /boot/config.txt /media/recovery/
(/boot/config.txt already contain disable_splash=1 line)
reboot

But it gives me an impression that it did not save any boot time as i thought. The splash lasts maybe 12 seconds and without it there is black screen and boot time is roughly same.

procount, 2020-01-11:
Correct.
It takes virtually no time to display the rainbow test screen. The Pi is not just sat there doing nothing whilst the splash screen is showing, it is still initialising.

closed 2020-01-13 by XECDesign

lurch, 2020-01-17:
Yup, IIRC the rainbow splash is displayed by the GPU when `start.elf` is loaded, and ISTR reading somewhere that it's actually only 4 pixels which are blended and scaled to fill the whole screen.

I don't _think_ I've ever seen the rainbow splash being displayed for a whole 12 seconds though :confused:  Is your SD card particularly slow?