raspberrypi/noobs #569: Adding AP

procount, 2020-01-21:
Question for @XECDesign & @maxnet particularly... :wink:
I want to add a wifi access point to PINN, so I'm mimicking a setup I have successfully working in Raspbian.
So far I have added the following buildroot (2015.2) packages: iw, iptables, dnsmasq and hostapd
and I'm just trying to get it going using cmdline scripts to make sure I've installed everything I need,
but I'm failing on my first step.
I'm executing:
`hostapd -B /mnt/hostapd.conf -P /tmp/hostapd.pid`

Here is my hostapd.conf
```
# cat /mnt/hostapd.conf
interface=wlan0
driver=nl80211
ssid=PINN
hw_mode=g
channel=6
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=password
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
```
and this is the result:
```
# hostapd -dd -B /mnt/hostapd.conf
random: Trying to read entropy from /dev/random
Configuration file: /mnt/hostapd.conf
rfkill: Cannot open RFKILL control device
nl80211: RFKILL status not available
nl80211: Using driver-based roaming
nl80211: TDLS supported
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: Disable use_monitor with device_ap_sme since no monitor mode support detected
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 2 iftype 3 (AP)
nl80211: Setup AP(wlan0) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0xe0df08 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0xe0df08 match=
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=0): [NULL]
nl80211: Could not configure driver mode
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=2)
netlink: Operstate: ifindex=2 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 2 iftype 2 (STATION)
nl80211: Teardown AP(wlan0) - device_ap_sme=1 use_monitor=0
nl80211 driver initialization failed.
hostapd_interface_deinit_free(0xe0c280)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0xe0c280)
hostapd_bss_deinit: deinit bss wlan0
hostapd_cleanup(hapd=0xe0c758 (wlan0))
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
hostapd_interface_free(0xe0c280)
hostapd_interface_free: free hapd 0xe0c758
hostapd_cleanup_iface(0xe0c280)
hostapd_cleanup_iface_partial(0xe0c280)
hostapd_cleanup_iface: free iface=0xe0c280
```
I don't appear to have rfkill and there is no config setting in buildroot to enable it. Is it necessary? It seems to continue without it.
But it looks like the nl80211 driver is not being initialised, yet I have the BR2_PACKAGE_WPA_SUPPLICANT_NL80211 [=y] setting enabled.
Will I need to go through the DBUS interface, like is used to enable the current wifi in NOOBS? I really have no idea how that all works.
Would appreciate any pointers....

lurch, 2020-01-21:
If you haven't already, it might also be worth asking on the Buildroot mailing list / forum ?
(I suspect running an AccessPoint might be one of the things that Buildroot is used for quite commonly? :man_shrugging: )

procount, 2020-01-21:
Thanks. I've searched again and found a project that sets up an AP on buildroot 2017. It doesn't seem to enable rfkill either. I discovered I needed to add AP mode to wpa_supplicant, but it still fails in the same way,

procount, 2020-01-21:
That would be great, thanks!

procount, 2020-01-23:
I am inspired 😃

closed 2020-01-23 by procount