raspberrypi/noobs #570: noobs/pinn translation language alignment

procount, 2020-02-06:
I was asked about such alignment. see https://github.com/procount/pinn/issues/352
I think the 'ar' code in NOOBS should be 'ae'.

lurch, 2020-02-06:
https://github.com/raspberrypi/noobs/pull/568#issuecomment-582983628 :wink: 

It gets kinds confusing because to do internationalisation & localisation "properly" you need separate settings for Language, Country and Keyboard settings (and then you also have dialects like pt_BR for Brazilian Portuguese).
But (to try and keep things simple) NOOBS conflates all of these into a single setting.

procount, 2020-02-06:
So maybe 'ar' is correct, then.
OK. I'll look to revert changes in PINN then. I think it's largely an internal issue anyway.

closed 2020-02-06 by procount