raspberrypi/noobs #571: Display trouble while installing Raspbian with NOOBS 3.3.0

neuralassembly, 2020-02-07:
When I try to install Raspbian to Pi 4 with NOOBS 3.3.0, 
I noticed that my display does not show anything.

When using with NOOBS 3.2.1, it has no troubles.

Therefore, I finished the installation using another display,
and then I re-connected Pi 4 to the display with trouble.

It still does not show the rainbow square, boot messages, etc.
However, after finishing the boot, it suddenly shows the desktop of Raspbian.

I think that there might be some trouble of the firmware of NOOBS.
Are there any solutions?

XECDesign, 2020-02-07:
@popcornmix, @timg236, @6by9, any idea?

XECDesign, 2020-02-07:
Do you have the same problem if you use a normal Raspbian image rather than installing through NOOBS?

procount, 2020-02-07:
@neuralassembly - Some info on the displays might help the list of experts above. What make & model of display, default resolution, maybe even the EDID file from the troublesome display?
Did you try pressing key '2' in NOOBS to change to hdmi_safe mode?

neuralassembly, 2020-02-07:
I tried the Raspbian image, and the same problem happened (i.e., no rainbow square, etc).

So, it would be a problem of Raspbian, not that of NOOBS.

Thank you for your suggestion.