raspberrypi/noobs #572: Enable dual HDMI output

meltwater2000, 2020-02-10:
In order to keep things simple for beginners it would be ideal if NOOBS was able to output on both HDMI ports.  This would avoid the failure where someone flashes their SD-Card and only sees the Rainbow screen because they used the "wrong" HDMI port (this just makes it look like the SD-Card is broken or firmware issue).  Even pressing 1-4 won't help with this.

Note: The HDMI furthest away will often need to be used if a slimline HDMI adaptor isn't available, so perhaps the default HDMI choice isn't ideal.

If possible dual-screen output would be a better solution, selection via numeric buttons or different choice of default HDMI.

procount, 2020-02-12:
I'm willing to experiment with this, but where do I start?
```
max_framebuffers=2
+BR2_PACKAGE_QT_GFX_MULTISCREEN=y
Add "Multi:" to QWS_DISPLAY + probably some other strange incantations
API to switch screens in QT?
```

lurch, 2020-11-29:
It's not quite "dual display" but I guess #606 contains a possible workaround for this.