raspberrypi/noobs #573: Align United Araba Emitates language file name to locale standards

bovirus, 2020-02-11:
@XECDesign 

I propose for **_recovery_ae.ts_** to rename this file as **recovery_ar_AE.ts** and modify it inside as

language="**_ar_AE_**" sourcelanguage="en"

and modify the **_recovery.pro_** to use **recovery_ar_AE.ts**

to follow the standard https://lh.2xlibre.net/locales/

XECDesign, 2020-02-11:
We don't have a `recovery_ae.ts`, but `language="ae"` in `translation_ar.ts` is probably wrong. Is there any reason to use `recovery_ar_AE.ts` instead of `translation_ar.ts` if we are not dealing with multiple variants of scripts and/or language?

bovirus, 2020-02-11:
Right. Now the file mname is recovery_ae.ts
Then the standard file naming is "recovery_"+"language code"+".ts"
Then in this case language file name should be "recovery_ar_AE.ts".

If you run lupdate recovery.pro with "recovery_ar.ts with language="ar" report error
If you run lupdate recovery.pro with "recovery_ar.ts with language="ar_AE" report NO error

lurch, 2020-02-11:
AFAICT, as we don't have multiple dialects, it should be okay to stick with just `translation_ar.ts` ?
I think we'd only want to move to `recovery_ar_AE.ts` if we had a UAE-dialect-specific version of the Arabic translation? (which seems extremely unlikely!)

It's much the same way that we "only" have a `translation_fr.ts` and not separate `translation_fr_FR.ts`, `translation_fr_CA.ts`, `translation_fr_CH.ts` etc.

On the other hand, I'm not a QT expert :wink:

bovirus, 2020-02-11:
@lurch 

Arabic United Arab Emitates for Is not a dialet. 
The Arabic has several declinatiation for different region where you used it.

Please read carefull the list that I posted.
https://lh.2xlibre.net/locales/

The standard report that if your use language for United Arab Emitates the language code is not "ar" but is "ar_AE"

If you have only French language you can use "fr" because is master language.
If you use French language for Belgium you should use "fr_BE".

But for Arabic you cannot use "ar" for United Arab Emitates because is not master language for Arabic.

XECDesign, 2020-02-11:
@bovirus, sorry, but nothing you're saying aligns with what's in the repo, what I'm seeing as output of lupdate or Qt's documentation. I'm happy to `update language="ar"` because that makes sense. But I'd need help from @procount or @lurch to confirm the rest.

bovirus, 2020-02-11:
@XECDesign 

I made the same request to @bovirus 

procount use these filename 

translation_ae.ts
translation_ast.ts

You sue for NOOBS

translation_ar.ts
translation_ast.ts

Both solution for me (lancode ar / ae /ast pinn or NOOBS) are out of standard.

In both situation (pinn or NOOBS) if you run now **lupdate recovery.pro** you got both (in pinn and NOOBS) two errors about ae/ast (pinn) or ar/ast (NOOBS) language code.

If you follow the standard https://lh.2xlibre.net/locales/ and make these changes in NOOBS

---- changes for Arabic

rename **_translation_ae.ts_** as **translation_ar_AE.ts** (to follow the scheme ltranslation_xxx.ts)
modify in **_translation_ar_AE.ts_** the string language="**_ar_AE_**" sourcelanguage="en"
modify **_recovery.pro_** file to use **recovery_ar_AE.ts**

---- changes for Asturian

rename **_translation_ast.ts_** as **translation_es_AST.ts** (to follow the scheme ltranslation_xxx.ts)
modify in **_translation_es_AST.ts_** the string language="**_es_AST_**" sourcelanguage="en"
modify **_recovery.pro_** file to use **recovery_es_AST.ts**

you can run **_lupdate recovery.pro_** without ANY errors bevcause language codes are OK.
Tested myself.

XECDesign, 2020-02-11:
That link relates to glibc and isn't directly related to Qt's locale string. Qt just needs the language code and everything else is optional. It determined the country (if it needs to) based on the language if it's not specified.

> If country is not present, or is not a valid ISO 3166 code, the most appropriate country is chosen for the specified language.

From what I gather, the core of the issue for you is that you get errors from lupdate?

What errors do you get? I don't get any either way.

5.14.1 is quite far from what's in buildroot.

bovirus, 2020-02-11:
When you run **lupdate recovery.pro** it didn't recognize ae / ar /ast as valid country code.
ar_AE and es_AST are recognized as valid country code.
And it match the documents that I posted,

XECDesign, 2020-02-11:
Can you paste the exact output?

bovirus, 2020-02-11:
Updating 'translation_ar.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
File .../noobs-master/recovery/translation_ast.ts won't be updated: it contains translation but the target language is not recognized
Updating 'translation_ca.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_de.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_es.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_eu.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_fi.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_fr.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_gl.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_hu.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_it.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_ja.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_ko.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_nl.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_no.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_pl.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_pt.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_ru.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_sk.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_sv.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_tr.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_zh_Hans_CN.ts'...
    Found 161 source text(s) (0 new and 161 already existing)
Updating 'translation_zh_Hant_TW.ts'...
    Found 161 source text(s) (0 new and 161 already existing)

procount, 2020-02-11:
For comparison, this is the output from PINN using `lupdate version 4.8.6`
```
Updating 'translation_ae.ts'...
    Found 639 source text(s) (0 new and 639 already existing)
Updating 'translation_ast.ts'...
    Found 639 source text(s) (0 new and 639 already existing)
Updating 'translation_ca.ts'...
    Found 639 source text(s) (0 new and 639 already existing)
```

XECDesign, 2020-02-11:
So `ar` is fine, but the looks of it. 

Some history on why `ast` is used - https://github.com/raspberrypi/noobs/pull/274#issuecomment-283051087

`es_AST` would be the Spanish with an invalid ISO 3166 country code.

bovirus, 2020-02-11:
"ae" and "ast" are wrong.
"ar" don't report error but is for Generic Arabic,
"ar_AE" is Arabic for United Arab Emitates then I beleive that it's better.

XECDesign, 2020-02-11:
Yes, `ae` is wrong. `ast` is wrong, but there is no alternative. `ar_AE` offers no advantage at this time.

closed 2020-02-11 by XECDesign

XECDesign, 2020-02-11:
For reference, if it comes up again: `ast` was added in Qt 5.0.0:

https://github.com/qt/qtbase/commit/3791d21f72a4c7035cd1db42c36dbd1dc7efd0f6

lurch, 2020-02-11:
> Yes, `ae` is wrong. 

Ironically, it looks like that error was introduced by @bovirus :laughing: 
https://github.com/raspberrypi/noobs/pull/453

bovirus, 2020-02-11:
"es_AST" is the right working fix for "ast".
"ar_AE" is the right working name for the contents of actual "translation_ar.ts" file. 
"ar" is Generic Arabic and "ar_AE" is Arabic for United Arab Emitates.
The contents of "translate_ar.ts" is Arabic for United Arab Emitates.
"translation_ar.ts" contains string not for generic Arabic but for Arabic for United Arab Emitates,

bovirus, 2020-02-11:
@lurch

The post was of THREE YAERS AGO.
I don't have so much experience as now in QT SDK and in the past no one say nothing different or prose different solutions.

procount, 2020-02-11:
I should probably change PINN back to using 'ar' then. 
At least it will be consistent with NOOBS and they can both be wrong. :smile:

AIUI, the current limitation is QT4.8 as it doesn't see those codes as errors, and doesn't even support 'ast', whereas QT5 has better support. So we can re-visit this again when we change to Qt5?

XECDesign, 2020-02-11:
We've gone around in circles enough, but one more time:

The format we're using here is ISO639[_ISO3166] . As documented here https://doc.qt.io/archives/qt-4.8/qlocale.html#QLocale-2

`es` is Spanish from an invalid country code, so Qt will just treat that as Spanish. That's wrong. You don't get an error because the country isn't important, but the language ends up being wrong.

We can use `ar_AE` when we have Arabic translations from other regions.

XECDesign, 2020-02-11:
Also, I can't find any reference to 'generic Arabic' or how it might differ from the Arabic written in UAE.

https://en.wikipedia.org/wiki/Varieties_of_Arabic
https://en.wikipedia.org/wiki/United_Arab_Emirates#Languages

lurch, 2020-02-12:
> "ar" is Generic Arabic and "ar_AE" is Arabic for United Arab Emitates.
The contents of "translate_ar.ts" is Arabic for United Arab Emitates.
"translation_ar.ts" contains string not for generic Arabic but for Arabic for United Arab Emitates,

@bovirus If you're really that concerned about the differences (I'm afraid I know absolutely nothing about Arabic myself), I suspect a "more appropriate" fix might be to update `translate_ar.ts` to use "Generic Arabic" rather than "Arabic for United Arab Emirates" ? :man_shrugging:

bovirus, 2020-02-12:
@lurch 

If the language contents in the file is "Generic Arabic" 

language name = translation_ar.ts
id = TS version="2.0" language="ar" sourcelanguage="en"

If the language contents in the file is "Arabic for United Arab Emitates" 

language name = translation_ar_AE.ts
id = TS version="2.0" language="ar_AE" sourcelanguage="en"

lurch, 2020-02-12:
Sorry, I meant if you're that bothered about the differences between "Generic Arabic" and "Arabic for United Arab Emirates", then perhaps a better fix for NOOBS is to **update the translations inside the `translate_ar.ts` file** to use "Generic Arabic" rather than "Arabic for United Arab Emirates" ? :man_shrugging: 

IMHO we should prefer to have "generic" language translations, rather than country-specific language translations.

commit 1f09ca44 (2020-02-11): translation_ar.ts: ae->ar

Fixes #573