raspberrypi/noobs #577: Shall README 'Setup' mention that Pi 4B may get video only in one HDMI port?

aszasz, 2020-03-23:
Once the Setup section of the README.md is thoughtful enough to say "Please note that in some cases it may extract the files into a folder, if this is the case then please copy across the files from inside the folder rather than the folder itself."

I think it would be equally worth to mention that "For Pi 4 Model B, it has been reported that switching to HDMI port closer to the power supply has solved the lack of video response"

That is because it took me long to try the swicth, as well as other reported here ( https://raspberrypi.stackexchange.com/questions/101842/noobs-stuck-at-rainbow-screen-raspberry-pi-4 )

(I also requested raspberrypi.org to mention it on the set-up guide)

lurch, 2020-03-23:
A side-effect of #572 not being done (yet?) :wink:

aszasz, 2020-03-24:
May I submit a pull request adding the followng line to README.md (it would be the first line of Setup section)?
**Raspberry Pi 4 Model B:** display must be connected to HDMI port closest to power jack.

JamesH65, 2020-03-24:
You can use either port, UNLESS you want 4kp60, when you need to use HDMI0.  I think that is the only limitation.

procount, 2020-03-24:
@JamesH65 - Yes, it's using the legacy driver. Tell me how to mirror the screen to both HDMI ports in Qt and I'll put it in PINN too.

JamesH65, 2020-03-25:
Mirroring on the legacy driver is possible I think, but not sure currently. A way to do it would be the same same bitmaps for each output, but that would encounter problem if you had two displays of differing resolution. But why mirror, don't you just use the attached display? Where ever it is connected? tvservice -l will list the attached displays and their ID's, so whatever you are using for final output (after QT) can just use that?

procount, 2020-03-25:
With NOOBS being written in QT, I think QT takes care of all the video output, which is managed by including various QT libraries and setting the output modes with help of the QWS_DISPLAY environment variable. I worked out how to alter this to output to both the default display (usually HDMI0) and VNC simultaneously, but I've not worked out how to output to dual HDMI with Qt (see also #572). 

I suppose for dual displays with the same resolution, it would be possible to use some external fb copy program, like is done for some of the smaller spi type displays, but I think using qt's facilities would provide a better solution - I think it should be possible, I just don't know how to do it.

procount, 2020-03-25:
@aszasz - Anyone can submit a PR. I don't see a problem with this suggestion to update the README, especially as it will help avoid user frustration. But it will be up to the maintainers whether it is ultimately merged or not.

lurch, 2020-03-25:
@procount I don't have time to look at this myself, but have you already looked at https://doc.qt.io/archives/qt-4.8/qt-embedded-displaymanagement.html ?

procount, 2020-03-25:
@lurch - Thanks, but I think I read that some time ago, which is why I added my comments at https://github.com/raspberrypi/noobs/issues/572#issuecomment-585258136, but I didn't get it to work.

lurch, 2020-04-09:
Fixed by #578

closed 2020-04-09 by lurch