raspberrypi/noobs #579: Display on HDMI furthest of USB ports (related to: #577 #578 #572)

AndyCapiau, 2020-03-28:
Related to issues: #577, #578, #572 

Usecase: downloaded Noobs 3.3.1. (offline version) from raspberry pi site, installed it on raspberry pi 4, 4gig model (sd card of 64 gig formatted to fat32).

Contrary to all issues I read here and online, my display only works on the hdmi port furthest of the USB ports (I believe your terminology is: not hdmi 0, but on hdmi 1 port). (I have the same issue on pinn by the way). 

The issue then is: 
- if I want to see the noobs boot menu, I have to connect my screen to the hdmi port furthest of the USB ports and all OS'es that boot from the menu, also display on that same (hdmi 1) port furthest of the USB ports.
- when I however connect my screen to the hdmi port closest to the USB ports, and boot my pi, let noobs just select my last os and boot that, then that OS is displayed on the hdmi 0 port (however I then never see the bootmenu... In reality it's displayed, but on the other hdmi port, so I never see the output, the OS boots, finds the screen on hdmi 0 and uses that as expected). 

I read that hdmi 0 is the more potent port and hdmi 1 (the one furthest of the USB ports) is less potent, so I kinda fear that if I use the bootmenu, I'm always forced to use the less potent port and that this will bring trouble down the road.

If I can somehow provide you guys with more info, please provide me the specific commands and I'll post the output here.
I mainly posted this, because it seems to go against a lot of the issues filed here and goes contrary to lots of the 'solutions' I read online. My 'solution' was to use the hdmi port furthest of the USB ports.

procount, 2020-03-28:
I think you have just misinterpreted the descriptions.
Hold your Pi4 such that the HDMI ports are towards you with the USB2/3 and Ethernet sockets to your right.
Starting at the left and moving towards the right, the first socket is the USB-C socket that is used for the power supply. Next to that is HDMI-0 which is the main video output used for NOOBS/PINN. Then there is HDMI-1, the camera socket, AV jack and then the USB-2/3 sockets.
So HDMI-0 is the one closest to the **USB-C** power socket, which is furthest from the 4x USB2/3 sockets.

All descriptions of which HDMI port to use that I remember refer to the **USB-C** power socket, which you may have thought meant the 4x USB sockets on the end of the Pi4.

AndyCapiau, 2020-03-28:
Ow, wow, sorry guys to take up your time. Thanks for the patient reply... that was a dumb mistake on my part. I'll close this thread. 

Love the software by the way. Helped me bigtime to install libreelec, lakka and raspbian multiboot. Works like a charm.

closed 2020-03-28 by AndyCapiau