raspberrypi/noobs #580: Format just a system

chenteparao, 2020-04-09:
Hi all, I am very new in NOOBS

I have three systems in my sd card, but I crash just one, can I only format that system through noobs or need start again with three?

procount, 2020-04-09:
You have several options, depending on your experience and inclination:
1. NOOBS does not have the facility to replace just one OS, so you would need to start again and install all 3 OSes again.
2. You could replace your single OS manually by downloading the install files for it and extracting the files over your crashed system. You then need to fix up the partition references manually by editing cmdline.txt and /etc/fstab. (There may be additional steps depending on which OS you are referring to, but since you installed with NOOBS, I doubt there will be any).
3. You could replace NOOBS with PINN, which is a compatible fork of NOOBS. This has the ability to backup, restore, replace and reinstall individual OSes. I wrote a post on how to do it somewhere, if only I could find it!

chenteparao, 2020-04-09:
Thank You man, I think I go for pinn.

lurch, 2020-04-09:
Closing issue, as this is something that NOOBS doesn't support (and it sounds like the user has a workaround).

lurch, 2020-04-09:
This is also a duplicate of #193  :wink:

closed 2020-04-09 by lurch

procount, 2020-04-09:
@chenteparao - Can't find my post, but this one is close enough -> https://github.com/procount/pinn/issues/108