raspberrypi/noobs #581: NOOBS reports insufficient space on 8GB SD Card

JinxDojo, 2020-04-10:
I followed the directions in "INSTRUCTIONS-README.txt" but was unable to install Raspbian due to insufficient space. It stated there was ~5100MB available but ~6000MB required. The card was formatted (as the instructions indicate) and only the contents of "NOOBS_v3_3_1.zip" was copied over--nothing else.

The card is 8GB from SanDisk. I had previously used it to successfully install a previous version of Raspbian using NOOBS (v1.9.0 I think). I'm not sure how all the repartitioning works, but if an 8GB card is no longer supported, the instructions/readme should probably be updated. Of course, ideally, it'd be great to get support back, but I understand if this is not possible. In that case, there should be a recommendation to use an imaging tool instead for 8GB cards (which I have done successfully using the same card).

procount, 2020-04-10:
Did you download the full version of NOOBS? This includes the install files for Raspbian, Raspbian-lite and LibreElec which wastes a lot of space.

On an 8GB card I suggest you just install NOOBS-LITE and install Raspbian from the internet. 16 GB SD cards are more the norme these days. 8 GB is a bit small for the full desktop version.

You can convert full NOOBS to NOOBS-LITE by deleting the /os/ folder and it's contents. Then add runinstaller to recovery.cmdline and reboot. This will delete anything you've already installed and repartition your card with more space for Raspbian. But you will need an internet connection to install Raspbian.

JinxDojo, 2020-04-10:
Thank you for the suggestion. I did download the full version of NOOBS (from the URL noted below). Personally, I'm all set at this point, but your suggestion should probably be put in the instructions/readme, which continues to claim an 8GB card will work. While it'd be nice to see a version that could still fit (maybe a "Raspbian only version"?), if that's not realistic, then the instruction/readme should probably state that "16GB cards are recommended and 8GB cards require NOOBS Lite (online installation requiring the Pi to have internet access) or alternatively using a disk imaging tool."

https://www.raspberrypi.org/downloads/noobs/
SHA-256: 649c22dd1e64d5d80884e804287505c4af11b6541817597ca7cac9aca7d9ccfc

hackercyclops12, 2020-04-27:
16 gigabyte is the best and recommended by rasberry pi themselves

andrum99, 2020-04-28:
> 16 gigabyte is the best and recommended by rasberry pi themselves

Nope - the official documentation recommends a minimum card size of 8GB:

https://www.raspberrypi.org/help/noobs-setup/2/

This is linked to from the NOOBS download page at https://www.raspberrypi.org/downloads/noobs/

lurch, 2020-04-28:
https://www.raspberrypi.org/documentation/installation/noobs.md now says "Your SD card will need to be at least 16GB for Full Raspbian, or at least 8GB for all other installs."

@JamesH65 @XECDesign should we try to get that same phrasing on https://www.raspberrypi.org/help/noobs-setup/2/ ?

JamesH65, 2020-04-28:
Yes, will ping the webteam.

James

hackercyclops12, 2020-04-28:
#https://github.com/raspberrypi/documentation/blob/master/installation/noobs.md
scroll down and you will see:

> Your SD card will need to be at least 16GB for full Rasbian

andrum99, 2020-04-28:
> #https://github.com/raspberrypi/documentation/blob/master/installation/noobs.md
> scroll down and you will see:
> 
> > Your SD card will need to be at least 16GB for full Rasbian

Yes, you are correct in that instance. However, the NOOBS download page linked to https://www.raspberrypi.org/help/noobs-setup/2/ which, until today, still said 8GB. This has now been corrected.

lurch, 2020-04-28:
> This has now been corrected.

Hugs all round :hugs: 

#582 is in the process of fixing this issue for the NOOBS README.

JinxDojo, 2020-04-30:
Thanks to everyone for addressing this discrepancy--it looks like it'll be
corrected on both the guides and the README/instructions file from now on;
hopefully, that will save some headaches!

Thanks again.

lurch, 2020-05-05:
Fixed in #582 and #586 

Thanks for bringing the issue to our attention.

closed 2020-05-05 by lurch