raspberrypi/noobs #583: Do not save WiFi credentials on SD card

seborama, 2020-04-28:
I have just found out that the NOOBS installer saved my WiFi credentials on one of the NOOBS partitions after I installed raspbian a few weeks ago.
I don't think this is good practice and I can't think of any good reason for doing that.

Could you please change the installer not to write WiFi credentials on NOOBS's paritions?

Just for clarity, this was done by NOOBS, this is **not** a file I created myself as documented in the README (https://github.com/raspberrypi/noobs#preconfiguring-a-wifi-network).

lurch, 2020-04-28:
> I don't think this is good practice and I can't think of any good reason for doing that.

Because it saves you having to re-enter them whenever you want to re-install an OS using NOOBS?

> I don't think this is good practice and I can't think of any good reason for doing that.

Just out of curiosity, what's the concern?

procount, 2020-04-29:
I imagine it's a question of storing the wifi passwords in plain text.
Maybe we should be using wpa_passphrase to store them in encrypted fashion?

XECDesign, 2020-04-29:
I don't think wpa_passphrase adds any security if you're at a point that somebody can get their hands on your wpa_supplicant.conf. Unless you're talking about a case where somebody only has a chance to glimpse at it and memorise it. Maybe my imagination is too limited, but I'm having trouble coming up with many practical scenarios where that happens.

If you have access to the if you know the SSID and the PSK, you can connect to the network without the passphrase. The passphrase is just easier to remember and enter than the PSK.

procount, 2020-04-29:
Maybe I misunderstood the use of wpa_passphrase (I've never used it).
I thought that having something like:
```
# network section generated by wpa_passphrase
network={
    ssid="MYSSID"
    psk=59e0d07fa4c7741797a4e394f38a5c321e3bed51d54ad5fcbd3f84bc7415d73d
}
```
was more secure than
```
network={
    ssid="MYSSID"
    psk="MYPASSWORD"
}
```
But I suppose it is just obfuscation. Whilst you don't get to know the SSID password, you can still connect if you have access to the psk.

andrum99, 2020-04-29:
wpa_passphrase adds a bit of obfuscation. It does provide a certain amount of protection from the case where the user has elected to use the same password for their wifi as for something else. This is a common weakness in how end-users make use of passwords.

If NOOBS used wpa_passphrase on all wifi passwords the user entered, then stored the calculated hash instead of the plaintext password into wpa_supplicant.conf, would that work? Is there any downside to that approach?

seborama, 2020-05-01:
Hi,
NB: I found another copy of the file under /etc on the raspbian installation that is universally readable.

procount, 2020-05-01:
Yep. NOOBS is no different to Raspbian in how it stores wifi credentials.

andrum99, 2020-05-01:
Ok then, so this should be closed as 'wont fix'.

lurch, 2020-05-01:
@seborama Can you provide a concrete example of why / how this causes a problem?

XECDesign, 2020-05-04:
Alright, closing for now, but will keep an eye on it.

closed 2020-05-04 by lurch