raspberrypi/noobs #585: Switching to composite for NOOBS display no longer works

andrum99, 2020-05-01:
Current NOOBS Lite (v3.3) is unable to switch between HDMI and composite, and is unable to output on composite even if I attempt to force the use of composite using a config.txt file with the required settings. I have tested this on a Pi 1 B rev 1 and Pi 4B 4GB rev 1.1. I expect this is also broken on Full NOOBS, but have not tested this.

With the Pi 1B, if the Pi is connected to a display via HDMI before the Pi is powered up, it will successfully switch between 1. HDMI preferred and 2. HDMI safe, but 3. and 4. give no display on any connector. If the Pi is not connected to an HDMI display before the Pi is powered up, the Pi fails to produce a display from any connector when keys 1,2,3 and 4 are pressed. The Pi appears to successfully set the HDMI video mode on 1. and 2., since the display reports the video mode as 640x480@60Hz, but the screen is blank black. On this display, no signal results in a blue screen.

With the Pi 4B, it is required to put enable_tvout=1 in config.txt to get composite output. With this in place, the Pi will not produce a display on the composite output. I did not test the HDMI output.

To check if this was a firmware issue, I replaced all of the firmware files on the NOOBS Lite SD card with those from a working Raspbian Lite installation. That Raspbian Lite, the latest version (2020-02-13), allows me to switch composite video output using config.txt parameters. Using those firmware files on the NOOBS Lite SD card got me some kind of output on the Pi 4B's composite port, however since I do not have the correct cable, all I get is garbled output. I believe the cable has ground and signal pins reversed. Nevertheless, this proves that composite is currently broken in NOOBS, and is caused by a firmware issue. Removing enable_tvout=1 on the Pi 4B gets the rainbow splash, then it gets stuck at a black screen. That is probably to be expected, since I'm not using matching fixup, start and kernel files (fixup and start from Raspbian Lite, kernel from NOOBS Lite).

See discussion with @procount on the forum at: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=272680

Kyuchumimo, 2020-12-13:
I confirm this issue, and in fact it is a persistent issue to this day (NOOBS v3.5.0).
I have a Raspberry Pi 2B.

ridgekuhn, 2021-07-11:
Installed latest NOOBS today and I have this issue on a 3B. Adding `display=3` to recovery.cmdline does not output NTSC to composite out. Additionally, setting `display_mode=3` in noobs.conf on /SETTINGS get wiped out after attempting to boot. If you open the file again, it will show `display_mode=` with no value.