raspberrypi/noobs #588: 28th release not compatible with new Pi 4 bootloader

parkerlreed, 2020-05-28:
https://downloads.raspberrypi.org/NOOBS/images/NOOBS-2020-05-28/NOOBS_v3_4_0.zip

Complains about recover4.elf not being up to date. I can't find this file anywhere online up to date to try and replace it.

XECDesign, 2020-05-28:
@timg236 why would the bootloader refuse to load an older start file?

timg236, 2020-05-28:
Probably because usb boot is selected

parkerlreed, 2020-05-28:
Yeah I'm trying to boot this from USB. Completely slipped on that on the initial report.

lurch, 2020-05-28:
> I can't find this file anywhere online

See https://github.com/raspberrypi/noobs/blob/master/BUILDME.sh#L178 :wink:

parkerlreed, 2020-05-29:
Aha! I had a hunch it was reusing one of the newer files but was never sure. That explains a bit.

I don't have it with me to test currently but feel free to close this out if it's known working.

timg236, 2020-05-29:
NOOBS isn't updatable and whilst you can probably copy start4cd.elf over recover4.elf you'll need update the other bootfs paritions plus cmdline.txt and fstab to use PARTUUID. However, the complexity was rather higher than just creating a new RPiOS install.

A future NOOBS will support USB once firmware is no longer beta.  I think this can be closed as 'wontfix'

procount, 2020-05-29:
I have a test version of PINN working with USB-BOOT, and will be working on being able to upgrade existing OSes. I may release it as a beta to get some feedback, before a proper release once the firmware is stable.

XECDesign, 2020-05-29:
Thanks everyone

closed 2020-05-29 by XECDesign

procount, 2020-06-19:
PINN 3.3.4.2 should now support eeproms that have been updated with the beta firmware for USB-BOOT. This also means PINN will boot from USB, **_BUT_** any installed OSes may have to be updated manually to support USB-BOOT for the moment.
NOTE: This is still considered BETA software.