raspberrypi/noobs #590: `marketing.tar` content

michaelbaudino, 2020-06-01:
Hi there :wave: 

I'm working on adding a new OS to NOOBS and I'd love to have a few informations regarding the expected content of `marketing.tar`.

1. the only used directory inside `marketing.tar` is `slides_vga`, right? ([source](https://github.com/raspberrypi/noobs/blob/master/recovery/mainwindow.cpp#L1583-L1595)) _(it looks like there has been some plan to also use `slides` on larger screens but the code has been commented for 7 years :grimacing:)_
2. the window will actually resize to the first image size ([source](https://github.com/raspberrypi/noobs/blob/master/recovery/progressslideshowdialog.cpp#L63-L66)) so what would be the **recommended** size for slides?

Thanks a lot in advance, any help is appreciated :pray:

lurch, 2020-06-01:
@procount probably has some good advice, as I believe he's been repackaging other OSes into "NOOBS format" for his PINN project :slightly_smiling_face:

procount, 2020-06-01:
IIRC, 387x292 was the most common size, but they can look a bit small on a Full HD screen. So you can go bigger, but I'd suggest to keep it under 600x440 so they fit on a standard VGA screen or HDMI safe mode. PINN uses a slightly different resizing method to NOOBS so it is a bit more forgiving of the various sizes used,

michaelbaudino, 2020-06-02:
Awesome, thanks for your answer @procount :raised_hands:

closed 2020-06-02 by michaelbaudino