raspberrypi/noobs #591: Build fails with "404: Not Found" for kernel image

ossguy, 2020-06-03:
When I run `./BUILDME.sh` it builds for about an hour but then I get this:

`
--2020-06-03 14:58:45--  http://sources.buildroot.net/linux-f16e91dad2af9d57aef477cc1f522040353849f5.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 2606:4700:20::681a:25, 2606:4700:20::ac43:4838, 2606:4700:20::681a:125, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|2606:4700:20::681a:25|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-06-03 14:58:45 ERROR 404: Not Found.
`

It looks like that image might have existed at one point (per the hash that was added in https://github.com/raspberrypi/noobs/commit/17e54d97858de81501afe8abfb611cc903b50975 ) but does not exist any longer.  Perhaps this hash needs an update?

lurch, 2020-06-04:
That's weird... IIRC buildroot only falls back to downloading source tarballs from http://sources.buildroot.net/ if the "primary download site" fails (this _occasionally_ happens when buildroot relies on a particular version of a package, but the team developing that package releases a new version and stops providing older versions). However that obviously only works for source-tarballs that buildroot has mirrored, and I **wouldn't** expect them to be mirroring raspberry pi linux kernel source tarballs :wink: 

So that suggests that buildroot tried downloading `http://sources.buildroot.net/linux-f16e91dad2af9d57aef477cc1f522040353849f5.tar.gz` because it was unable to download commit `f16e91dad2af9d57aef477cc1f522040353849f5` from `git://github.com/raspberrypi/linux.git` - however it seems that commit _does_ still exist in the repo https://github.com/raspberrypi/linux/commit/f16e91dad2af9d57aef477cc1f522040353849f5 (which is also tagged as [raspberrypi-kernel_1.20200205-1](https://github.com/raspberrypi/linux/releases/tag/raspberrypi-kernel_1.20200205-1) )
:confused:

procount, 2020-06-04:
I often had this problem with buildroot and some of the rpi repos.
Before running BUILDME.sh, download all sources first by using:
```
cd buildroot
make source
```
If a package fails, go to GitHub, select the appropriate commit and copy the link to download it. It will be in .zip format, so change the link to .tar.xz and download. Check the downloaded file is exactly as buildroot expects it. Sometimes you will need to add an 'rpi-' prefix. Copy it to buildroot/dl. Repeat to get all packages. Then you can run `BUILDME.sh`

lurch, 2020-06-04:
> download all sources first by using...

Just tried that myself in a fresh clone of the NOOBS repo, and it all went fine and downloaded almost a gigabyte of tarballs, so perhaps the original problem was just a temporary failure? :shrug:

procount, 2020-06-04:
Downloading the repos for linux and rpi-firmware nearly always fail for me. I don't know what's special about those. I reckon you must have a super-fast pipe to github from PI Towers! :wink:

ossguy, 2020-06-04:
It looks like the issue was related to the first comment - the download from sources.buildroot.net was attempted because of an issue with getting the commit.  Reviewing further up in my log file, the issue with getting the commit ended up being that the Git command failed due to running out of disk space on my end.  So a bit unfortunate that the error wasn't clearer (which is why I'll leave this open in case there's an easy fix to be done there), but at least I've been able to solve it.

Now my task is to determine how to build the "Offline and network install" version of NOOBS, since it appears that what's produced by `./BUILDME.sh` is just the "Network install only" version.

lurch, 2020-06-04:
>  I reckon you must have a super-fast pipe to github from PI Towers! :wink:

:rofl: Working from home at the moment (so only on home broadband), although my VirginMedia 108Mbps connection isn't too bad :wink: 

> the Git command failed due to running out of disk space on my end.

Whoops! :wink: 

> (which is why I'll leave this open in case there's an easy fix to be done there)

It'd be a buildroot-level "fix" rather than a NOOBS-level "fix" so I'll close this issue.

> Now my task is to determine how to build the "Offline and network install" version of NOOBS

IIRC the only difference is the contents of the `os` folder?

closed 2020-06-04 by lurch

lurch, 2020-06-04:
> It's only necessary to download that commit anyway, not the whole repo.

Yup, but unfortunately you need to download the whole repo if you want to be able to pluck out just a single commit - limitation in `git` AFAICT. 'shallow clones' exist, but they only get you the latest commit, not an arbitrary commit.

I looked into this quite a while ago, and IIRC some projects worked around this by setting up a git-to-svn mirror on the server, and then using the svn mirror to download a single commit to the client, as SVN _does_ allow you to "clone" a single arbitrary revision.