raspberrypi/noobs #594: how noobs to select different boot partition?

CXianRen, 2020-06-10:
hi, I review the code of recovery, and I notice that: it use syscall to reboot the rpi and jump to another system. 
code:
 reboot_part = getFileContents("/run/reboot_part").trimmed();
    ::syscall(SYS_reboot, LINUX_REBOOT_MAGIC1, LINUX_REBOOT_MAGIC2, LINUX_REBOOT_CMD_RESTART2, reboot_part.constData());

and I try to compile  "noobs" of myself, which mean I want to reboot and boot and another system from current system. but I failed. what I miss ?

matthuisman, 2020-06-10:
does Pi4 now support reboot {part}?
Wasn't it not possible due to the Sd card speed change?

CXianRen, 2020-06-11:
I am  using RPI4 B 2G 2018 version
and the noobs can work in my pi.
@procount  
@matthuisman 
Yes, when I reboot 2 , it still reboot from partition from parition 1

matthuisman, 2020-06-11:
@CXianRen 
read this:
https://github.com/raspberrypi/noobs/issues/551

CXianRen, 2020-06-11:
I get it. 

The fix to the issue I thought this was was introduced in 241-4, so yes, it's definitely a part of Raspbian now.

Sorry, I completely forgot about this, but you can't reboot into different partitions when the uSD interface is running on 1.8V (UHS). You need to boot with sdhci.debug_quirks2=4 in cmdline.txt for it to work, but it will sacrifice SD speed (I don't know how much).

AIUI, the issue is that once the card is in UHS mode, we can't get out of it without resetting power, losing the content of the reset register (which contains the reboot partition number). Due to reasons I don't quite understand, we can't continue in UHS mode and must switch to 3.3V to properly initialise the card.

@matthuisman  @procount  @lurch  thank you so much

closed 2020-06-11 by CXianRen