raspberrypi/noobs #596: Support  installation of Raspberry Pi OS on F2FS root filesystem

jimis, 2020-06-29:
[F2FS](https://en.wikipedia.org/wiki/F2FS) has been designed on the basis of log-structured filesystems, with the goal of performing optimally on managed flash media like SD cards, minimizing writes and extending their lifetime. It has seen some usage on Android phones, and it's constantly being improved with features that Raspberry Pi would very much like to have, like online compression.

Currently Raspberry Pi OS root partition can be converted to F2FS using a time-consuming process:
* Mount SD card with Linux PC
* Back up root partition
* Make F2FS filesystem on the SD card
* Restore backup to the new partition
* Update `fstab` and boot command line on the SD card

This is a dangerous and write-intensive process. In order to avoid it, it would be nice if installation could be done directly on a F2FS root partition.

ret394, 2020-11-24:
Not a great or good idea for NOOBS. This one complicates it for NOOBS maintainers.

You can just find the clues and build a filesys from scratch.

kdm6389, 2022-07-09:
backup exiting partition/
tar --create --preserve-permissions --gzip --verbose --file=/md0/RasPi.Backup.FS.tar.gz --acls --selinux --xattrs --exclude=/proc --exclude=/tmp --exclude=/mnt --exclude=/dev --exclude=/md0 --exclude=/sys --exclude=/media --exclude=/run --exclude=/boot --exclude=/var/log --exclude=/home/pi --exclude=/var/cache/apt/archives --exclude=/usr/src/linux-headers* --exclude=/home/*/.gvfs --exclude=/home/*/.cache --exclude=/home/*/.local/share/Trash  --one-file-system --record-size=715776K /

this will crate FS tarball /md0/RasPi.Backup.FS.tar.gz
/md0 is external location like USB drive 

take your sdcard and usb on the linux pc, format sdcrad parttion new F2FS 
then restore 
sudo tar -xvpzf /path/to/RasPi.Backup.FS.tar.gz -C /mount/sdc2 --numeric-owner

may this help you