raspberrypi/noobs #597: Support for ExFAT filesystem?

michaelbaudino, 2020-07-19:
Hello,

Recalbox core team member here :wave: 

We are switching one of our partition filesystem to ExFAT in our next release (due in a few weeks), but from what I can see [in this piece of code](https://github.com/raspberrypi/noobs/blob/master/recovery/multiimagewritethread.cpp#L747-L775) there's no way to ask NOOBS to create an ExFAT filesystem from our `partitions.json` file :sob: 

Are there any plans to support this? Or at least any custom filesystem creation command? If not, which approach do you think we should consider?

Thanks a lot in advance for your help.

PS: I've tried a few things already to reformat and create the filesystem on first boot, but all my attempts failed:
* `parted` refuses to resize the extended partition that contains the logical partition mounted as `/`
* `fdisk` is made for humans and thus its input is highly inconsistent and not reliably usable by a script
* `sfdisk` doesn't seem to support GPT

michaelbaudino, 2020-07-20:
Just to be clear, I'd be happy to provide a PR to support this (let me know if you want it!), but it doesn't sound like a good solution to me because it's not backward compatible: people who own (or receive from a manufacturer) a NOOBS SD-card created with an older version would not be able to install our OS.

maxnet, 2020-07-20:
>Or at least any custom filesystem creation command?

You can run your own arbitrary commands as post-installation script.
Windows does that as well.

michaelbaudino, 2020-07-20:
Awesome idea, I'll try that :raised_hands:
Thanks a lot for the hint @maxnet :pray:

lurch, 2020-07-20:
I hope you don't mind me asking, purely out of curiosity...

> We are switching one of our partition filesystem to ExFAT in our next release

Why? :)

I've just googled and found https://en.wikipedia.org/wiki/ExFAT#Adoption and to answer your original query NOOBS doesn't (yet) use a 5.4 kernel :wink:  (Although apparently it [does support FUSE](https://github.com/raspberrypi/noobs/commit/ff4060743aeaeb942f1884dd6914ea2905d842c3), which I didn't realise!)

lurch, 2020-07-20:
> it doesn't sound like a good solution to me because it's not backward compatible

We've had similar backwards-compatibility problems in the past, and that's why current versions of NOOBS download [os_list_v3.json](https://github.com/raspberrypi/noobs/blob/master/recovery/config.h#L25) with [older versions](http://downloads.raspberrypi.org/) still being supported.

michaelbaudino, 2020-07-21:
> I hope you don't mind me asking, purely out of curiosity...
>
>
> Why? :)
>
> I've just googled and found https://en.wikipedia.org/wiki/ExFAT#Adoption and to answer your original query NOOBS doesn't (yet) [...]

Ahah sure, feel free to ask :hugs:

This partition is the one where users can store their media content (roms (games), music, movies, …). Some users have **a lot** of content.

Since we are targeting non-technical users, we need this partition to be readable/writable on Windows, thus we excluded EXT4. We also excluded FAT32 due to technical limitations (partition size, file size, access rights, no journaling, …). We've been supporting NTFS for some time too, but have a few issues with it (mounting is slow, partitions are randomly not detected, …). It's probably an issue on our side, since NTFS seems quite mature, but we never had time to properly track that kind of not reproductible bugs. Finally, we recently tried ExFAT, which is well supported on both Windows (natively) and Linux, and is quite solid/stable, from what we can see.

Anyway, we may move back to NTFS at some point (since NTFS optimizes space better than ExFAT, iirc), but for now, we're quite happy with ExFAT broad support and performance.

-----------------

Regarding the driver, we are using [this FUSE-based driver](https://github.com/relan/exfat) which is actually [provided by Buildroot](https://github.com/buildroot/buildroot/blob/master/package/exfat/Config.in). We may switch to native Linux 5.4 support, but we didn't bump to 5.4 on all architectures yet (only on Pi4 at the moment, actually).

michaelbaudino, 2020-07-21:
@maxnet:

> Regarding partitions.json:
> Set "empty_fs" to true to tell NOOBS you don't want the partition formatted.
> "partition_type" to whatever MBR partition id exFAT uses (2 hex digits).

It worked :raised_hands: 

Well, actually, I also had to add `"filesystem_type": "unformatted"` too, otherwise NOOBS tries to run `mkfs`, which is not what we want.

Actually, I was expecting `"filesystem_type": "unformatted"` to be enough (_i.e._ that it would imply `"empty_fs": true`), but no :man_shrugging: With only `"filesystem_type": "unformatted"`,  NOOBS does not create the filesystem (_i.e._ it does not run `mkfs`, as we would expect), but it still tries to download a tarball and populate the (not-existent) filesystem.

Let me try to make this more ~~&nbsp;clear&nbsp;~~ nerdy.

Current behaviour pseudo-code:

* _if_ partition is defined in `partitions.json` _then_
  * create partition
  * _if_ `filesystem_type` is not `unformatted` _then_
    * create filesystem (`mkfs`)
  * _if_ `empty_fs` is not true _then_
    * populate filesystem with tarball content

Shouldn't it rather be:

* _if_ partition is defined in `partitions.json` _then_
  * create partition
  * _if_ `filesystem_type` is not `unformatted` _then_
  	* create filesystem (`mkfs`)
    * _if_ `empty_fs` is not true _then_
	  * populate filesystem with tarball

Anyway, thanks a lot for your help :pray: