raspberrypi/noobs #599: VNC has issues when you do not renew/change the screen resolution

rugk, 2020-07-22:
we have VNC support (ref. https://github.com/raspberrypi/noobs/issues/116) and [even docs for it](https://www.raspberrypi.org/documentation/remote-access/vnc/), solving [an old question](), but there seems to be a big issue with the system resolution:

If you go through the setup as intended (in my case only after enabling direct capture mode and rebooting, though that may be unrelated) you need to **go to raspi-config and reset the screen resolution to something non-default**.
I seem to be not the only one experiencing this as that workaround has been described here:
https://desertbot.io/blog/headless-raspberry-pi-4-remote-desktop-vnc-setup#step-11-important-change-the-default-screen-resolution

Though as no issue has been reported for this, I'd guess I'll open one here. You can find the full STRs in the blog post there.

andrum99, 2020-07-22:
That's not a bug - it's actually working as intended. If there is no screen display locally, then VNC has nothing to connect to. In that case you can either use raspi-config as you describe to force the HDMI output to be active, or alternatively run the VNC server in virtual mode.

rugk, 2020-07-22:
If you enable "boot to desktop [and autologin]" there is a screen to display – always. As such VNC works.

Also, you only need to apply that workaround once (change the screen resolution) – even if you previously changed that already when connected via HDMI or so. It works flawlessly when you reboot all the next times.

procount, 2020-07-22:
Apart from the reference to #116, which seems irrelevant, what has this got to do with NOOBS?

lurch, 2020-07-23:
This is an issue in Raspberry Pi OS rather than NOOBS, so closing.

closed 2020-07-23 by lurch

rugk, 2020-07-23:
Hmm okay, so where is the issue tracker for the Raspberry Pi OS?
I found https://github.com/raspberrypi/Raspberry-Pi-OS-64bit, but that only applies to the 64bit version – and, well you onyl offer 64bit now, so hmm??

lurch, 2020-07-23:
Issue tracker for the documentation is https://github.com/raspberrypi/documentation/
Issue tracker for Raspberry Pi OS itself is either https://github.com/RPi-Distro/repo or https://github.com/RPi-Distro/pi-gen (unless @XECDesign can suggest somewhere more suitable?)

rugk, 2020-07-23:
Okay, moved to https://github.com/RPi-Distro/repo/issues/186