raspberrypi/noobs #608: Recovery Wallpaper

duckida, 2020-11-16:
Hi. I checked the wallpaper([/recovery/wallpaper/png](https://github.com/duckida/noobs/blob/master/recovery/wallpaper.png)) and noticed it was **from 2013!!!**. I have created a more modern one:
[image]

Can we please use this new one? It will be the first step in making NOOBS consistent with Raspberry Pi OS!

duckida, 2020-11-16:
@lurch Please See

duckida, 2020-11-16:
Feel free to make your own recovery wallpapers and submit them

lurch, 2020-11-17:
@duckida It's @XECDesign 's decision to make, not mine :wink:

XECDesign, 2020-11-17:
These kinds of changes would need to go through the (very busy) design team. We are currently only making basic maintenance changes to NOOBS to keep it working with newer models, so I don't think the wallpaper is really an issue that's likely to get looked at.

duckida, 2020-11-17:
@XECDesign Please See #610

closed 2020-12-02 by XECDesign