raspberrypi/noobs #610: NOOBS UI Redesign

duckida, 2020-11-17:
Hi. I have looked through NOOBS and found that many of the UI components are not consistent with Raspberry Pi OS. We need to change the icons, wallpaper, and color scheme for it to be consistent. I created an issue to change the wallpaper(#608), but it got the **wontfix** label.

XECDesign, 2020-11-17:
Like I mentioned in that other issue, there are no plans to do any non-essential work on NOOBS. And as you say, it's not just a matter of changing the wallpaper. If we were to give it a refresh, it would be a full reskin, upgrade to newer Qt and so on, which would be a significant time commitment.

For context, NOOBS was developed back in the day when the pi was more susceptible to SD card corruption. Since it (almost) never wrote to the boot partition itself, it was a way for users to start fresh when their sd card got trashed without having to flash the card from another computer.

Since that is no longer an issue, it's being phased out in favour of rpi-imager and will be discontinued at some point in the future. I hope that helps explain why it looks a bit neglected.

If you're interested in a more actively maintained fork of NOOBS, please take a look at PINN: https://github.com/procount/pinn

lurch, 2020-11-17:
ping @procount - Are there any parts of PINN that @duckida could help you with?

procount, 2020-11-17:
I don't really have any plans to update the UI at the moment. PINN already has user customisable wallpaper, background colours and selectable themes (from QTs limited selection). But if @duckida has any suggestions, I do consider appropriate PRs.
At some point it would be nice to update PINN to a newer Qt and buildroot version, but that does appear to be technically challenging based on the attempts so far.

Help is always welcome on the slideshows during OS installation, translations, and even improving the translation arguments (like someone promised a long time ago  :wink: )

ret394, 2020-11-24:
Absolutely not enough time.

>(like someone promised a long time ago 😉 )

If it was me, mention the username. Though I doubt I'd take up frivolous pursuits like that.

closed 2020-12-02 by lurch