raspberrypi/noobs #612: Few supported_models strings please (CM3 Lite & Pi-400)

matthuisman, 2020-11-19:
These are the strings used for matching NOOBS / PINN compatible OS's (supported_models)

Can anyone confirm Raspberry Pi Compute Module 3 Lite  would be **Pi Compute Module 3 Lite Rev**
And that the PI400 is using **Pi 4 Model B Rev**

procount, 2020-11-19:
"Raspberry Pi 400 Rev 1.0"

matthuisman, 2020-11-19:
i was about to say that no OS's would match that... but "Pi 4" would actually match that 👍

matthuisman, 2020-11-19:
just found there is a CM3+ Lite as well.
would that be  **Raspberry Pi Compute Module 3 Plus Lite Rev**

lurch, 2020-11-19:
I don't have any CM4s, but I think I have a CM3 Lite - if I remember I'll try digging it out tonight and see what `/proc/cpuinfo` and `/proc/device-tree/model` says.
**However** IIRC the "Lite" compute-modules don't actually "know" that they're "Lite" modules, as they just see the (external) SD card attached to exactly the same pins as the (internal) eMMC would be.

procount, 2020-11-19:
Scanning the DTB files for strings yields the following list, but it is not complete (e.g. no Pi 400) as I think the firmware can make some adjustments.

Raspberry Pi Zero W
Raspberry Pi Model B
Raspberry Pi Model B+
Raspberry Pi Model B
Raspberry Pi Compute Module
Raspberry Pi Zero
Raspberry Pi Zero W
Raspberry Pi 2 Model B
Raspberry Pi 2 Model B rev 1.2
Raspberry Pi 3 Model B
Raspberry Pi 3 Model B+
Raspberry Pi Compute Module 3
Raspberry Pi 4 Model B
Raspberry Pi Compute Module 4

lurch, 2020-11-19:
Sometimes things have to be deliberately obfuscated before products are launched, for obvious reasons :wink:

closed 2020-12-11 by matthuisman

lurch, 2020-12-12:
Ahh! I spent a tedious evening a couple of weeks ago swapping SD cards, HDMI cables, power cables, USB cables, etc. back and forth, and managed to collate this list (forgot to post it earlier). I couldn't find my CM3+, and I don't have a CM4 yet (and I couldn't be bothered to flash the eMMC on a CM1). Details were copied by hand from the HDMI output (i.e. not copy'n'pasted from SSH) so there _may_ be transcription errors.

Revision | Model
--- | ---
0x0002 | Raspberry Pi Model B Rev 1
0x0008 | Raspberry Pi Model A Rev 2
0x000e | Raspberry Pi Model B Rev 2
0x0010 | Raspberry Pi Model B Plus Rev 1.2
0x0012 | Raspberry Pi Model A Plus Rev 1.1
0x0015 | Raspberry Pi Model A Plus Rev 1.1
0x900092 | Raspberry Pi Zero Rev 1.2
0x9000c1 | Raspberry Pi Zero W Rev 1.1
0x9020e0 | Raspberry Pi 3 Model A Plus Rev 1.0
0xa02082 | Raspberry Pi 3 Model B Rev 1.2
0xa020a0 | Raspberry Pi Compute Module 3 Rev 1.0
0xa22042 | Raspberry Pi 2 Model B Rev 1.2
0xb03111 | Raspberry Pi 4 Model B Rev 1.1
0xc03130 | Raspberry Pi 400 Rev 1.0
0xd03114 | Raspberry Pi 4 Model B Rev 1.4