raspberrypi/noobs #613: supports_usb details

procount, 2020-11-24:
Could someone please explain/remind me of the difference and reasoning behind `support_usb_boot` and `supports_usb_root` ?

My understanding is as follows:

`supports_usb_root` allows an OS to boot off an SD card but store the root partition on a USB drive. As NOOBS uses partuuids to reference partitions on USB, this means the **OS must support partuuids**.

`supports_usb_boot` allows NOOBS and the OS to boot from USB. So this means the **OS must support USB boot.** Does this also imply `supports_usb_root` must also be true, since the root will also normally be on the USB? 

What if the user boots NOOBS off USB, but selects to install root to SD card (unlikely but I guess it could be possible - I've not checked that possibility). Would this require `support_usb_root` ? One would think not, but the fstab might reference the boot partition, which would be on USB and therefore referenced by partuuid, so maybe it should be true also?

I'm asking this because I now have an OS that could boot off usb and store the root on usb, but it cannot support partuuids. Normally this would be excluded, but I see no reason why it couldn't be accommodated if the user accepts that the USB drive must always be /dev/sda. I thought of setting "supports_usb(boot,root)=true" but add "use_partuuids=false" to indicate it can use the device references, but it is not backwards compatible. So maybe I need to add new "supports_sda_boot" and "supports_sda_root" parameters. 

Anyone got an opinion on the best way to do this?

lurch, 2020-11-24:
I've not looked into the NOOBS source code, so can't answer most of your questions, but with specific reference to "if the user accepts that the USB drive must always be /dev/sda." - as you may know, the way Linux allocates `/dev/sda`, `/dev/sdb` etc. to different USB drives is on a first-come-first-served basis [1], so if you have two USB flash drives (or a USB flash drive and a USB hard disk) connected, there's no way of guaranteeing which will be `sda` and which will be `sdb` (which is presumably why NOOBS references USB partitions by partuuid rather than `/dev/sdX1`).
This means if you _did_ allow a user to install an OS to USB using `/dev/sda1` and `/dev/sda2`, if the user ever happened to have an extra USB drive plugged in at the same time as booting up NOOBS, there's a chance they might end up with `/dev/sda` pointing at their "extra" drive, and `/dev/sdb` now pointing to their NOOBS drive, at which point the NOOBS-installed OS would be unable to fully boot because it wouldn't be able to find the appropriate root partition at `/dev/sda2`.
Of course you may well have known all that already... :wink: 

So while you're obviously free to do whatever you like in PINN, I think it's unlikely that this would ever be supported in NOOBS, given how easy it would be for the user to accidentally "break" it by just plugging in an additional USB drive before bootup. Kind of goes against NOOBS's simplicity and reliability ethos IMHO.

[1] I think you might be able to override this with custom UDEV rules, but I dunno if that's something that NOOBS supports.

procount, 2020-11-24:
Yes, I knew all that and the rationale for using partuuids to avoid the device reference problems.
(And obviously I'm asking this for PINN, but asking it here for the expert advice only as it will never get done in NOOBS ) :wink:

I'm not sure reading the source code would help answer the questions anyway, as I am asking about the use cases or scenarios that led to the decision to use these parameters. 

I've already testing it in PINN with an appropriate warning message about the consequences, but my solution is not backwards compatible with NOOBS or older versions of PINN, so I'm thinking of changing it to the "supports_sda_(boot,root)" variant, unless anyone can think of anything better. Just asking in case I missed something.

lurch, 2020-11-24:
Is adding "supports_sda_(boot,root)" to PINN easier than asking the OS concerned to support partuuids? :shrug:

procount, 2020-11-24:
In this case yes, as it will not support partuuids.
I do wonder if this will help install some of the existing OSes that currently don't support root on USB & those where I have no influence - I have some testing to do there! But the onus will be on the user to understand the consequences of this, hence it needs warning messages.

ret394, 2020-11-24:
> Kind of goes against NOOBS's simplicity and reliability ethos IMHO.

5 days of silence for the humour inherent in this seemingly innocuous and serious statement.

----

>not backwards compatible with NOOBS

Don't worry about it, I'll advice non tech-savvy folks to keep away from it in a harsh tone. So they are scared like a horror movie. For example - something a normal person finds scary when watching alone or in a cemetary. 

PS - a cemetary is a cold place. Carry gloves and warm water. Lights are not important, and no one disturbs you when you're typing away in the beautiful moonlight.

PS - You must not be the overly imaginative type. Gravestones don't move on their own or appear to open weird non-Euclidean portals, there is a scientific explanation.

---

supports_usb_boot  that's the only one that should be used in case you are writing a tutorial for prepping the Raspi for USB boot.

And please update your Raspi articles. Use a VCS if you need it.

ret394, 2020-11-24:
Is this PINN any good?

XECDesign, 2022-01-13:
@procount Can this one be closed?

closed 2022-01-13 by procount