raspberrypi/noobs #614: NOOBS+

duckida, 2020-12-01:
Hello! I have made some changes to NOOBS! You can view them at [duckida/noobsplus](https://www.github.com/duckida/noobsplus). These changes make NOOBS up-to-date and easier to understand!

lurch, 2020-12-01:
In order to avoid any possible confusion between the official NOOBS releases, and your own custom version, would you mind choosing a different name please?

Also, the start of the NOOBS licence says:
```
Copyright (c) 2013, Raspberry Pi All rights reserved.

Redistribution and use in source and binary forms, with or without modification, are permitted provided that the following conditions are met:

Redistributions of source code must retain the above copyright notice, this list of conditions and the following disclaimer.
```
which I think means you can't just retroactively claim "Copyright (c) 2020, N's Apps All rights reserved.". You can claim copyright over your modifications, but you have to also acknowledge the copyright of the original code.

It's great to see your enthusiasm though :slightly_smiling_face:

duckida, 2020-12-01:
Can I change the copyright to 2020 at least?

lurch, 2020-12-01:
https://opensource.stackexchange.com/questions/5484/how-to-use-mit-license-in-a-project might answer your questions?

closed 2020-12-02 by XECDesign