raspberrypi/noobs #615: Noobs download

Jayson380, 2020-12-02:
what causes "unable to expand NOOBS v3_5_0.zip

procount, 2020-12-02:
If you are just using standard Windows file manager to expand it, try installing 7zip and use that instead.

Jayson380, 2020-12-02:
Im using Macbook 13"

> If you are just using standard Windows file manager to expand it, try installing 7zip and use that instead.

Im using Macbook.

lurch, 2020-12-02:
https://www.raspberrypi.org/documentation/installation/installing-images/README.md suggests some alternative "unzip" programs.

@XECDesign Does NOOBS now use ZIP64 as well?

procount, 2020-12-02:
That page omits the fact that Raspberry Pi Imaging Utility can also be used to write NOOBS zip files.

lurch, 2020-12-02:
Feel like submitting a docs PR @procount ? :wink:

procount, 2020-12-02:
I'll see if I can fit it in :wink:

XECDesign, 2020-12-02:
> @XECDesign Does NOOBS now use ZIP64 as well?

Not sure off the top of my head, but the full archive probably does. Can investigate further if The Unarchiver doesn't work and the checksum has been verified.

closed 2022-01-13 by XECDesign