raspberrypi/noobs #616: How to change OS boot queue (NOOBS, Raspberry Pi 3)?

Goldenweiser, 2020-12-15:
How to manage the queue of operating systems in NOOBS on Raspberry pi 3? 
I need to change the priorities of boot to another operating system after each reboot

procount, 2020-12-15:
NOOBS will only automatically boot the last OS you chose by default. 
You can use the autoboot.txt file and set boot_partition to the OS partition number you want to boot, but this bypasses NOOBS.

PINN has an additional option of a "sticky default" which allows one OS to be the preferred OS to boot, but you can temporarily boot another, knowing that next boot it will boot the sticky selected OS by default again.

lurch, 2020-12-15:
> I need to change the priorities of boot to another operating system after each reboot

If you know what you're doing, I guess you could mount NOOBS's settings partition and edit the `default_partition_to_boot` setting in `noobs.conf` ? https://github.com/raspberrypi/noobs/blob/master/recovery/bootselectiondialog.cpp#L191-L195

Goldenweiser, 2020-12-15:
> NOOBS will only automatically boot the last OS you chose by default.
> You can use the autoboot.txt file and set boot_partition to the OS partition number you want to boot, but this bypasses NOOBS. [...]

Fantastic! Can you tell me, please, how change "sticky default" to another OS? In my case i have Windows IOT and Raspbian. And want to boot another OS after rebooting reaspberry
Or
Tell me, please, how to automatic change autoboot.txt ? Please tell, can Noobs execute the script and change autoboot.txt automatically? Where to add this script?

procount, 2020-12-15:
> how change "sticky default" to another OS? 

That is only applicable for PINN - an alternative form of NOOBS - just for your information. I think you can do what you want with NOOBS.

> how to automatic change autoboot.txt ? Please tell, can Noobs execute the script and change autoboot.txt automatically? Where to add this script?

NOOBS does not have the capability to run your own scripts. You need to write it your self and run it in your own OSes.
So in Raspbian, write a script when it boots to change the next OS to run. And then in that OS, write another script to change it back again, or whatever sequence you want to do.

You can do what Lurch suggested and change the setting in noobs.conf, or change the setting in autoboot.txt. In both cases you need to set the partition number of the boot partition of the OS you want to boot next. You can find this in installed_os.json.

If you are running NOOBS from an SD card, then autoboot.txt should be created in the first RECOVERY partition (/dev/mmcblk0p1). Create a line such as:
```
boot_partition=6
```
where 6 to boot from the first installed OS.

noobs.conf is found on the settings partition (/dev/mmcblk0p5). Find the line that says `default_partition_to_boot=6` and change the 6 (or whatever it is) as above.

installed_os.json is also found in the same place as noobs.conf. Look for a partitions section in the OS section you want. It will contain a list of partitions, such as:
```
"/dev/mmcblk0p6",
"/dev/mmcblk0p7"
```
Choose the last digit of the first partition (e.g. 6 above) to use as the partition number to boot.

Good luck if you need to do this in Windows IoT - I have no clue how to do that.

XECDesign, 2020-12-17:
General support queries are best addressed on the forum.

closed 2020-12-17 by XECDesign

lurch, 2020-12-17:
Seems like this discussion got "migrated" to https://github.com/procount/pinn/issues/450 :wink: