raspberrypi/noobs #617: I don't think these are the right files

JohnB17, 2020-12-27:
I couldn't figure out how to use this repo, but I found noobs at the below url, if you could could you upload the downloads on this site instead?

procount, 2020-12-28:
This repo contains the source files to build NOOBS. 
If you just want to use NOOBS, download the files from the https://www.raspberrypi.org/downloads/noobs/ URL you found, or from https://downloads.raspberrypi.org/

lurch, 2020-12-28:
https://github.com/raspberrypi/noobs clearly says "The latest official release of NOOBS can be downloaded from http://downloads.raspberrypi.org/NOOBS_latest "  ?

I'm going to close this issue but feel free to follow up with additional questions if things still aren't clear.

closed 2020-12-28 by lurch