raspberrypi/noobs #618: Trigger NOOBS recovery mode remotely via ssh

schklom, 2020-12-31:
Hi everyone !

My Raspberry Pi 4 (installed NOOBS on it) is at home in another country, and I want to remotely trigger a factory reset on it.
I accidentally deleted some system files, but luckily I stopped the deletion quickly and now I can still access it remotely.
I obviously can't hold down Shift while starting up the device, and I need to be able to access it by ssh after.

Does anyone know how to make this happen ? Is there a script I could run to trigger a factory reset (with ssh access) on NOOBS ?

Many thanks :)

procount, 2020-12-31:
Unfortunately, I think you might be out of luck with NOOBS.
It does support VNC, but unless you have already enabled VNC to be port-forwarded through your firewall (which is not advisable due to security considerations), or can enable such port-forwarding remotely through ssh , you won't be able to use it.
Even if you manage to use VNC, unless your system image already has some remote access system enabled (VNC or SSH), you won't be able to enable these through NOOBS because it does not support SSH.

PINN supports ssh, so you might get further with that, but I think someone would need to do an "Apollo 13 ground simulation" first to make sure it would work, because you risk losing access to your system altogether.

If you can access your system through ssh at the moment, IMHO your best bet would be to try and identify the files you have deleted and try to restore them directly.

lurch, 2021-01-01:
As @procount says, NOOBS itself doesn't support SSH, and I agree with his comment that "your best bet would be to try and identify the files you have deleted and try to restore them directly." (so I'll close this issue as "wontfix")

If you want to try that, the original tarballs that NOOBS installs can be found at https://downloads.raspberrypi.org/raspios_armhf/
(but bear in mind that some of those files will have been overwritten with newer versions during an `apt upgrade`, so be careful).
If you need further help you'll probably find there's a lot more helpful users at https://www.raspberrypi.org/forums/ than there are watching the NOOBS github issue-tracker :wink:

closed 2021-01-01 by lurch