raspberrypi/noobs #619: Build issue in the applying patches to binutils step

misang3, 2021-01-15:
Just cloned and trying to build an image for the raspberry pi.
I did a' make raspberrypi_defconfig' followed by 'make' and hitting this error every time. Any insights on what could be off? I'm on a Linux (Ubuntu 16) VM on an OS X host.

>>> host-binutils 2.24 Patching

Applying 001-fix-enable-install-libiberty-flag.patch using patch: 
patching file libiberty/configure
patching file libiberty/configure.ac

Applying 002-dont-segv-on-initial-instructions-overflow.patch using patch: 
patching file bfd/elf-eh-frame.c

Applying 120-sh-conf.patch using patch: 
patching file configure
Hunk #1 succeeded at 3793 (offset 2298 lines).
patching file configure.ac
Hunk #1 succeeded at 1129 (offset 417 lines).
patch: **** Can't rename file configure.ac to '' : No such file or directory
Patch failed!  Please fix 120-sh-conf.patch!
package/pkg-generic.mk:146: recipe for target '/mnt/hgfs/projects/raspberrypi/noobs/buildroot/output/build/host-binutils-2.24/.stamp_patched' failed

procount, 2021-01-15:
I've never needed to do `make raspberrypi-defconfig`. Maybe that's overwritten something, like .config?
And you don't run `make`, but run BUILDME.sh instead.

misang3, 2021-01-15:
Thanks for the prompt response. I'm seeing the exact same issue when i use BUILDME.sh and without the `make raspberrypi-defconfig` as well

procount, 2021-01-15:
The latest patches are to build it on Ubuntu 20.04, so it might be a good idea to upgrade to that.
The previous version was for Ubuntu 14.04 (and maybe 16 - I've never tried it), so if you can't upgrade to Ubunutu 20.04, checkout the previous version and try building that.

lurch, 2021-01-16:
```
Applying 120-sh-conf.patch using patch:
patching file configure
Hunk #1 succeeded at 3793 (offset 2298 lines).
patching file configure.ac
Hunk #1 succeeded at 1129 (offset 417 lines).
patch: **** Can't rename file configure.ac to '' : No such file or directory
Patch failed! Please fix 120-sh-conf.patch!
```
I'm not a `patch` expert, but I can't see anything in https://github.com/raspberrypi/noobs/blob/master/buildroot/package/binutils/2.24/120-sh-conf.patch that would make it want to rename a file? :confused: 

`'/mnt/hgfs/projects/...` are you mounting a filesystem path from outside your VM to inside your VM? I wonder if that might be causing problems? :man_shrugging:

closed 2022-01-13 by XECDesign