raspberrypi/noobs #620: CM4 Compatibility

NumberOneGit, 2021-01-30:
I'm trying to edit the config.txt file by adding "dtoverlay=dwc2,dr_mode=host" to enable the USB ports on the IO board to work in NOOBS, which works in Raspberry Pi OS and Ubuntu but does nothing in NOOBS.  Is there a way to get this to work?

procount, 2021-02-01:
@XECDesign - You only need to add CONFIG_USB_DWC2=y to make NOOBS compatible with the CM4.

sridharsaravanan, 2022-06-20:
gpu_mem=16
start_file=recovery.elf
fixup_file=fixup_rc.dat
enable_uart=1
dtoverlay=dwc2,dr_mode=host
otg_mode=1

[pi4]
max_framebuffers=2
start_file=recover4.elf
fixup_file=fixup4rc.dat
enable_uart=1
dtoverlay=dwc2,dr_mode=host
otg_mode=1

now it work add last two line