raspberrypi/noobs #623: Screen resolution can not be set with vc4-kms-v3d

audas, 2021-03-03:
Newest driver vc4-kms-v3d  and latest OS distribution and firmware remove the ability to set the resolution via the Screen resolution on the main drop down, and also via the raspi-config.

Happy to provide information about the issues if anyone interested.

lurch, 2021-03-03:
I assume you mean Raspberry Pi OS, rather than NOOBS? (NOOBS is basically just the OS "installer")

audas, 2021-03-03:
Oh ok - I tried raising this on raspberry/linux but they just closed the thread. I am unsure where to raise it sorry - it deals with both the firmware and operating system. The new firmware when dealing with the new driver is handled by the kernel - so the old firmware people have no interest - but the new kernel people don't think its their issue either - or more specifically not being able to set the resolution is a problem.

maxnet, 2021-03-04:
>Newest driver vc4-kms-v3d and latest OS distribution and firmware remove the ability to set the resolution via the Screen 
>resolution on the main drop down

Suggest that you are a little bit more specific what "main drop down" of what program no longer works.

"Pi icon" -> "Preferences" -> "Screen Configuration" ?
Think that lives in this repository: https://github.com/raspberrypi-ui/arandr

audas, 2021-03-04:
> Suggest that you are a little bit more specific what "main drop down" of what program no longer works.
>
> "Pi icon" -> "Preferences" -> "Screen Configuration" ?
> Think that lives in this repository: [...]

Yes - this is the one. Also if you go into raspi-config there is normally the ability to set resolution there as well - this is also removed.

I will post this on the above link thanks.

popcornmix, 2021-03-05:
> Turns out there is no support for changing the resolution with the new driver. Ok. Fair enough.

Don't really know what you mean by that. "Pi icon" -> "Preferences" -> "Screen Configuration" (internally arandr)
is a standard linux tool for changing display resolution and refresh rate. I used it regularly for fkms and kms drivers and it works as expected.

maxnet, 2021-03-05:
Apparently Simon thinks it is broken as well though: https://github.com/raspberrypi-ui/arandr/issues/1

maxnet, 2021-03-05:
@popcornmix

Broken for me as well.
With fkms I can change my screen to 800x600 through arandr:

[image]

With kms after selecting new resolution and pressing apply, no signal:

[image]

/sys/class/drm/card0/card0-HDMI-A-2/edid in case relevant: [edid.zip](https://github.com/raspberrypi/noobs/files/6090625/edid.zip)

popcornmix, 2021-03-05:
@maxnet that really is a different issue.
You are able to set the screen resolution, but a specific one (800x600) is not producing the correct timings on kms vs fkms.
An issue https://github.com/raspberrypi/linux/issues would be more suitable.

maxnet, 2021-03-05:
>You are able to set the screen resolution, but a specific one (800x600) is not producing the correct timings on kms vs fkms.

That was not the case.
Occurred on multiple resolutions, and my monitor is known to give "Input not supported" instead of "No signal" if it receives unsupported timings.

Anyway.
Problem is present in the kernel you ship with your January 2021 RPI OS image, but no longer in 5.10.20 which you get when you "sudo rpi-update".

So maybe that would solve the TS problems as well.

popcornmix, 2021-03-05:
You don't need rpi-update for 5.10 kernel. apt contains that.

audas, 2021-03-05:
For me this is on the latest kernel, latest os with kms. I can change the driver in the config, reboot and it dissapears, reappears.  (Did a full update upgrade).

I am able to use arandr via python calls to extract screen resolutions. 

Not sure if tvservice is used anywhere but this no longer works at all with new driver.

maxnet, 2021-03-05:
>I am able to use arandr via python calls to extract screen resolutions.

I am assuming you mean xrandr here.

>Not sure if tvservice is used anywhere but this no longer works at all with new driver.

What happens if you type "arandr" on the terminal?
Nothing starts? No console output either?

audas, 2021-03-05:
Ok that brings it straight up - showing the GUI with screen/s and resolution options.

And also you are correct regarding python - 

```
from Xlib import display
from Xlib.ext import randr
```

maxnet, 2021-03-05:
>Ok that brings it straight up - showing the GUI with screen/s and resolution options.

And it did not do that when you tried to start it through the menu?

audas, 2021-03-05:
Screen configuration used to appear - now screen saver appears (I can't remember what iterations this was) - here is a url which shows how it used to appear on the panel.

https://pimylifeup.com/raspberry-pi-screen-resolution/

If you notice further down it also shows the raspi-config tool being used - this no longer appears either. 

The raspi-config tool appears to show CEA in that image to my mind it may have been using the tvservice -l (I have no idea and very limited experience on these things).

closed 2022-01-13 by XECDesign